pimoroni/pimoroni-pico #207: Micropython RTC set_unix() not working and RTC not holding time across power down

hardillb, 2021-09-18:
I've been trying to sync the RTC to the current time and I'm having problems.

I've been using the Thonny console since this appears to set the current board time to the computer time when connected. I tried pasting the following script into the console.

As you can see the `set_unix()` returns `True` but does not update the rtc values returned after `update_time()` is called.

```
from pimoroni_i2c import PimoroniI2C
from breakout_rtc import BreakoutRTC
import time
PINS_PLASMA = {"sda": 20, "scl": 21}
i2c = PimoroniI2C(**PINS_PLASMA)
rtc = BreakoutRTC(i2c)
rtc.set_24_hour()
rtc.update_time()
print(rtc.string_time())
print(rtc.string_date())

time.localtime()
rtc.set_unix(time.time())
rtc.update_time()
print(rtc.string_time())
print(rtc.string_date())
True
00:00:25
01/01/2000
(2021, 9, 18, 18, 14, 41, 5, 261)
True
True
00:00:25
01/01/2000
```

If I use the `rtc.set_time()` with suitable values it works:

```
>>> rtc.set_time(20,23,18,6,18,9,2021)
True
>>> rtc.update_time()
print(rtc.string_time())
print(rtc.string_date())
True
18:23:44
18/09/2021
```

Secondary to this, when I disconnect the board from the USB port and attach to a USB PSU the RTC appears to have lost it's stored time.

Using a Plasma 2040 with the v0.2.6 release (MicroPython with Pimoroni Libs + Adafruit Blinka + Adafruit PlatformDetect)

Full code of `main.py` that is starting at epoc after setting RTC as above:

```
import plasma
from plasma import plasma2040
from pimoroni import RGBLED, Button
from pimoroni_i2c import PimoroniI2C
from breakout_rtc import BreakoutRTC
import time
 
PINS_PLASMA = {"sda": 20, "scl": 21}
 
i2c = PimoroniI2C(**PINS_PLASMA)
rtc = BreakoutRTC(i2c)
 
if rtc.is_12_hour():
    rtc.set_24_hour()
 
if rtc.update_time():
    print(rtc.string_time())
    print(rtc.string_date())
 
NUM_LEDS = 60
LOW = 32
MED = 64
HIGH = 128
BRIGHTNESS = [LOW,MED,HIGH]
BRIGHTNESS_LEVEL = 0
 
button_brightness = Button(plasma2040.BUTTON_A)
 
led = RGBLED(plasma2040.LED_R, plasma2040.LED_G, plasma2040.LED_B)
led.set_rgb(0, 0, 0)
led_strip = plasma.WS2812(NUM_LEDS, 0, 0, plasma2040.DAT)
 
led_strip.start()
 
rtc.enable_periodic_update_interrupt(True)
 
while True:
    RED = [0]*NUM_LEDS
    GREEN = [0]*NUM_LEDS
    BLUE = [0]*NUM_LEDS
    t = time.localtime()
 
    if rtc.read_periodic_update_interrupt_flag():
        rtc.clear_periodic_update_interrupt_flag()
          
        rtc.update_time()
        hour = (rtc.get_hours() % 12) * 5
        RED[hour] = BRIGHTNESS[BRIGHTNESS_LEVEL]
        RED[hour + 1] = BRIGHTNESS[BRIGHTNESS_LEVEL]
        RED[hour + 2] = BRIGHTNESS[BRIGHTNESS_LEVEL]
        RED[hour + 3] = BRIGHTNESS[BRIGHTNESS_LEVEL]
        RED[hour + 4] = BRIGHTNESS[BRIGHTNESS_LEVEL]
        GREEN[rtc.get_minutes()] = BRIGHTNESS[BRIGHTNESS_LEVEL]
        BLUE[rtc.get_seconds()] = BRIGHTNESS[BRIGHTNESS_LEVEL]
 
        for i in range (NUM_LEDS):
            led_strip.set_rgb(i, RED[i], GREEN[i], BLUE[i])
         
        if button_brightness.read():
            BRIGHTNESS_LEVEL += 1
            BRIGHTNESS_LEVEL %= 3
     
    time.sleep(1)
```

hardillb, 2021-09-18:
OK, I assume I need to call `rtc.set_backup_switchover_mode()` to enable the battery backup and allow the RTC to maintain the time, but I have no idea what value I should pass to this.

hardillb, 2021-09-18:
OK, looking at the desktop code I've assume `rtc.set_backup_switchover_mode()` should be passed `3` for level_switching_mode.

So the only outstanding issue is that `rtc.set_unix()` doesn't  look to work

Gadgetoid, 2021-09-18:
This might explain the `set_unix()` issue:

https://github.com/pimoroni/pimoroni-pico/blob/37209cd0c005c8c186fa63ab1c06ec714f2f525a/drivers/rv3028/rv3028.cpp#L385-L401

It looks like the unix timestamp is completely independent from the rest of the RTC... hooookaay! That's a gotcha. I think we might need a bit of documentation to cover that.

hardillb, 2021-09-18:
yeah, some docs for how to enable from new (`rtc.set_backup_switchover_mode()`) and that unix time != "real" time would be useful.

Thank.

closed 2024-06-11 by thirdr