pimoroni/pimoroni-pico #210: Pico Wireless Pack : cannot find network

bryht, 2021-10-03:
I bought this module
[image]

attach them like this:
[image]
[image]

and flash the uf2 with newest version: https://github.com/pimoroni/pimoroni-pico/releases/tag/v0.2.6

but when I try the example code : https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pico_wireless/cheerlights.py

it cannot get wifi connect, always return status 255, rather than 3.
then I tried get all the networks it returns 0.

[image]
[image]

Anyone may knows what the problem is? thanks in advance.

Gadgetoid, 2021-10-04:
This *could* be a soldering/electrical issue but I'm not sure how. I've run a couple of tests locally and I can list networks.

Might be worth noting that it finds only 2.4GHz networks and not 5GHz ones, so if your area somehow doesn't have any 2.4GHz networks that might also explain the lack of them.

bryht, 2021-10-04:
> This _could_ be a soldering/electrical issue but I'm not sure how. I've run a couple of tests locally and I can list networks.
>
> Might be worth noting that it finds only 2.4GHz networks and not 5GHz [...]

Hi Gadgetoid, thanks for the quick response, I double checked, I have 2.4Hz networks, really don`t know if it is an electrical issue, but when I run the code, the light has a strange color, seems blue,red,green together, do you know if that indicate something?
[image]

Gadgetoid, 2021-10-04:
It's fairly normal for the LED to just sort-of do its own thing when it's not being actively driven to a known state.

Could you take some photos of the Pico header soldering? From your side-on photo some of the joints look a little dry, but it's hard to tell. The solder should ideally form a little cone, covering the copper ring around each pin, rather than a bulge at the end of the pin.

bryht, 2021-10-04:
Hi Gadgetoid, that is quick 👍 

here are the Pico side, do you mean by this? can you tell any issue? actually I didn`t solder them, it is like this when I bought it.
[image]
[image]

bryht, 2021-10-04:
> It's fairly normal for the LED to just sort-of do its own thing when it's not being actively driven to a known state.
>
> Could you take some photos of the Pico header soldering? From your side-on [...]

Hi Gadgetoid, that is quick 👍

here are the Pico side, do you mean by this? can you tell any issue? actually I didn`t solder them, it is like this when I bought it.
[image]
[image]

Gadgetoid, 2021-10-04:
Ah! The header looks fine. It was just the shallow angle in the first picture that made it look funny.

Might be worth also firmly pressing the Pico and Pico Wireless together to make sure they're connected- though it seems like they are communicating at least.

I'm not sure what else could be wrong. 🤔

Gadgetoid, 2021-10-05:
Have run this by support, you should drop an email to support@pimoroni.com since they'd be able to replace it if it's a hardware fault.

closed 2021-10-05 by Gadgetoid