pimoroni/pimoroni-pico #212: Pico display thermometer.py degree symbol

trlaing, 2021-10-06:
Hi,
Great wee bit of kit! I am trying to display the temperature with the degree symbol before C, but cannot get it to work. I have tried the following, but all just show random looking characters instead of the degrees symbol:

`display.text("{:.2f}".format(temperature) + chr(176) + "c", 3, 3, 0, 3)`
`display.text("{:.2f}".format(temperature) + u"\u00B0" + "c", 3, 3, 0, 3)`
`display.text("{:.2f}".format(temperature) + "\xb0Cc", 3, 3, 0, 3)`

Gadgetoid, 2021-10-06:
The built-in LCD font is currently printable ASCII-only, chars 32-127. I tend to just use "C" to work around this.

helgibbons, 2021-10-07:
You can change the scale of the built in font - see https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/pico_display#text

closed 2022-07-11 by helgibbons