pimoroni/pimoroni-pico #219: Possible pin conflict

knightinlet, 2021-11-13:
For awareness and to save anyone else hours of investigation. Not sure if it's a bug or a bad combination on my part.

If using the pico breakout garden and RTC (rv3028) and breakout_colourlcd240x240 in the rear SPI slot combination then the display does not display as the default pin interrupt for RV3028 is 22 and the rear SPI CS is also pin 22.

knightinlet, 2022-01-25:
Hi,

Apologies I thought I had responded to share the knowledge. My bad.

At the time I finally figured that if I supplied another value to the constructor to prevent it defaulting to 22 it would solve my problem even if I wasn't using the interrupt feature. 

I was using the C++ libraries, but agree something in the docs might be useful for both implementations to help others in the future.

Gadgetoid, 2022-01-25:
I wonder if interrupt should default to -1 and only set up if supplied. Might avoid this in future. It's far from the only board that has an interrupt argument.

knightinlet, 2022-01-25:
That seems like a great design pattern to enable the feature only if needed. 

Would have saved me hunting for something I hadn't appreciated was activated outside of the i2c pins. 

That said was a great learning experience so thanks 😃

knightinlet, 2022-01-27:
What I should have said is that I passed RV3028::PIN_UNUSED as part of the constructor. Possibly default it to that and if you need interrupts you supply the PIN number as part of the constructor.

Gadgetoid, 2022-01-27:
Agreed.

Actually in practice there's virtually no point supplying the interrupt pin to the driver at all since it doesn't do anything with it. I might outright remove this parameter and merge some examples of interrupts actually being useful.

To actually *use* an interrupt in MicroPython you basically have to set it all up again: https://github.com/pimoroni/pimoroni-pico/issues/169

And I'm pretty sure (although I don't think I've played with pin interrupts) in C++ things are not much different. Interrupts are too messy to handle in the library internals.

Gadgetoid, 2025-05-28:
Closing as very, very stale- not sure I ever *did* anything about this, but it hasn't cropped up again 👀