pimoroni/pimoroni-pico #220: MicroPython Plasma can't set white LED for RGBW

zurzat, 2021-11-14:
set_rgb method seems to be written to accept white LED value:
`WS2812::set_rgb(uint32_t index, uint8_t r, uint8_t g, uint8_t b, uint8_t w, bool gamma)`

However, when trying to use this method in Python the compiler complains:
```
led_strip.set_rgb(0,0,0,0,128)
TypeError: extra positional arguments given
```

Ideally, the set-hsv method should also accept a white value.

helgibbons, 2021-11-15:
Are you using `rgbw=True` when you init your LED strip?

If you're using RGBW APA102s your init should look something like this:
```
import plasma
from plasma import plasma2040

LEDS = 30
FPS = 60

led_strip = plasma.APA102(LEDS, 0, 0, plasma2040.DAT, rgbw=True, color_order=plasma.COLOR_ORDER_GRB)
```

https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/plasma

zurzat, 2021-11-15:
Hi,
Yes, using NeoPixels WS2812. Following code is how I'm creating the led_strip object:
```
import plasma
from plasma import plasma2040
NUM_LEDS = 5*30
led_strip = plasma.WS2812(NUM_LEDS, 0, 0, plasma2040.DAT, rgbw=True, color_order=plasma.COLOR_ORDER_GRB)
```

Everything else works fine, can set colours etc.
Just can't specify the white value. The method won't accept more than 4 arguments.

Cheers.

dmptrluke, 2021-12-03:
I'm really surprised this basic feature isn't implemented at all, quite disappointing.

Gadgetoid, 2021-12-03:
> I'm really surprised this basic feature isn't implemented at all, quite disappointing.

I know everyone thinks I'm an infallible, great and perfect shiny golden god who never makes mistakes but that's only mostly true. I am occasionally distracted from being omniscient and miss something. Probably because I haven't had enough coffee. Coffee gives me power.

A fix for this is working its way through CI: https://github.com/pimoroni/pimoroni-pico/pull/226

zurzat, 2021-12-03:
@dmptrluke - it is implemented but there was a bug. 
Fantastic news that @Gadgetoid has implemented a fix :)
I'll try it ASAP!

Gadgetoid, 2022-01-25:
This *should* be fixed in version 0.3.2 - https://github.com/pimoroni/pimoroni-pico/releases/tag/v0.3.2

closed 2022-01-25 by Gadgetoid