pimoroni/pimoroni-pico #228: Using optical flow sensors simultaneously

roaldarbol, 2021-12-10:
I am really excited that the optical flow sensors (PMW3901 & PAA5100) are now supported on the Pico. I was wondering if it is possible to use two sensors simultaneously? The ["normal" PMW3901 library lets you set arguments](https://github.com/pimoroni/pmw3901-python/blob/master/examples/motion.py) including slots and ports, which opens the option of having 2 sensors recording simultaneously which I've used before, but it's not clear from the Pico [example](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/breakout_pmw3901/motion.py) how to achieve this. I'm not a wiz' in `.cpp`, but it seems that it might be possible from [the source code](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules/breakout_pmw3901/breakout_pmw3901.cpp), but I'm not sure how. Any ideas?

Gadgetoid, 2022-01-25:
Looks like we're lacking on the documentation front here.

The line:

```python
flo = FlowSensor()
```

Can take arguments to specify the SPI bus. You can - in theory - connect multiple sensors to the same SPI bus and nominate a different chip-select pin for each.

If you keep the default SPI bus, that's as simple as (the below are the pins for the Front/Back slots on Breakout Garden):

```python
flo1 = FlowSensor(cs=17)
flo2 = FlowSensor(cs=22)
```

You can specify different SPI pins altogether if you're wiring directly to a different bus:

```python
flo = FlowSensor()
```

Is equivalent to:

```python
flo = FlowSensor(
    cs=17,
    sck=18,
    mosi=19,
    miso=16,
    interrupt=20
)
```

roaldarbol, 2022-01-31:
Awesome, thanks! I'll look forward to testing it out and let you know how it pans out!

Gadgetoid, 2022-03-01:
There seems to be a maybe possibly undocumented way to control the LEDs but it's unimplemented in this library at the moment. See: https://github.com/pimoroni/pmw3901-python/issues/4

Looks like we don't define `BG_SPI_FRONT` or `BG_SPI_BACK` for MicroPython. That should probably be fixed!

You can add them at the top of your code like so for now:

```
BG_SPI_FRONT = 0
BG_SPI_BACK = 1
```

Note that they include the chip-select pin, so I think the init is just:

```python
flo = FlowSensor(
    slot=BG_SPI_FRONT
)
```

That cryptic `TypeError` definitely feels like the one I ran into. I'll have to try the same fix. Good grief, it's only through blind luck with a recent library that I have the slightest clue what it might be. Good timing, I suppose!

roaldarbol, 2022-03-01:
Haha, brilliant! The `slot` solution works - great!
Just ran a bit more, and when it crashes on me, the error is VERY similar looking (line 37 again being `get_motion`):
```python
Traceback (most recent call last):
  File "<stdin>", line 37, in <module>
TypeError: can't convert tuple to float
```

I think the reason I might have a bad feeling about the readings comes down to implementation - and they are definitely off for the back sensor. Currently I'm testing with:
```python
while(True):
    delta = flo1.get_motion()
    delta2 = flo2.get_motion()
    if delta is not None and delta2 is not None:
        x = (delta[0] + delta2[0]) // 2
        y = delta[1]
        z = delta2[1]
        tx += x
        ty += y
        tz += z
        print("A Relative, rel: x {}, y {}, z {} | Absolute: tx {}, ty {}, tz {}".format(x, y, z, tx, ty, tz))
    time.sleep(0.05)
```
The front sensor seems much more sensitive - I guess because I run them sequentially and it's first in line? Do you happen to have an idea of how it could be achieved to `get_motion` synchronously from them?

roaldarbol, 2022-03-16:
Was about to comment no worries - but wow, that was proper expediting! Really appreciate it! I'll give it a spin tomorrow and report back🤞🏼

roaldarbol, 2022-03-17:
Tested out, and it seems to have solved the issues. 👍  
As for my differences in readings, it seems it's due to a faulty or damaged sensor rather than code. Guess I'll just have to stock up a few. ;-) Thanks for the help!

closed 2022-03-17 by roaldarbol

Gadgetoid, 2022-03-17:
Whew truly a lucky coincidence I’d encountered this issue! I’ll get it merged and released post haste.