pimoroni/pimoroni-pico #230: RTC for Interstate75

carlskii, 2021-12-16:
Hi, I've been playing with the new interstate75 board. However, I notice it doesn't include an RTC which makes the clock examples somewhat limited unless its tethered to a computer. Do you have a recommend RTC for interstate75 and maybe an example on how to use it ?

Gadgetoid, 2021-12-16:
The RP2040 already has an RTC built-in. See: https://github.com/raspberrypi/pico-examples/blob/master/rtc/hello_rtc/hello_rtc.c

carlskii, 2021-12-16:
Thanks - so how would I get interstate75 to retrieve/maintain the correct time after it's booted ?

Gadgetoid, 2021-12-17:
My uh... unothadox solution was - albeit using MicroPython (where it's done automagically) - to allow the time to be set via a host computer, then use a battery to relocate the panel while powered on.

The time will be saved in the RTC during a power-off, but wont advance, so a quick swap to battery works quite well. Granted this isn't very robust!

If you want the RTC to keep ticking during a loss of power (or while powered down) you'd have to use something like the RV3028 with a little coin cell backup - https://shop.pimoroni.com/products/rv3028-real-time-clock-rtc-breakout wired using our Qw'st adapters and cables like so:

[image]

https://shop.pimoroni.com/products/stemma-qt-qwiic-to-breakout-garden-adapter
https://shop.pimoroni.com/products/jst-sh-cable-qwiic-stemma-qt-compatible?variant=31910609813587

Or add some controls to make manually setting the time in the built-in RTC easier- granted that'll lose time when power is lost.

ZodiusInfuser, 2023-01-20:
Closing this as it appears the question was answered. If it was not, please reopen.

closed 2023-01-20 by ZodiusInfuser