pimoroni/pimoroni-pico #236: Provide update method binding from ws2812 to micropython object.

shishu94, 2022-01-05:
In order to have a more fine grained control over the led strip refreshes, It would be great to have access to the update() method in the ws2812 driver directly from the MicroPython object.

Gadgetoid, 2022-01-05:
What's the use case?

If these LEDs aren't refreshed continuously then they're susceptible to noise. You cannot realistically, for example, update *once* and expect that pattern to say lit for any period of time without some noise creeping in and messing up the first few LEDs.

Alternatively you could be angling for something like "flip" where you can update a back buffer without the results showing, and call "flip" to replace the front buffer when your prepared back buffer is ready. IE: to avoid the difference between your updates and the screen refreshes causing tearing?

In the latter case... yeah it's probably easier just to expose `update` and caveat that it really should be called in a loop for best results.

shishu94, 2022-01-05:
My initial use case is exactly the dual buffer problem. Don't want the new color pattern to start showing before everything is set. 
I would also like to be able to have a variable refresh rate. Increase to 30 fps or more during animations and 2-5 fps during quiet periods. But that might be doable with the current API.

Gadgetoid, 2022-01-06:
I really should have added double-buffering into the library! I think exposing `update` could indeed solve your problem in the interim.

Gadgetoid, 2022-01-25:
This should be available now in v0.3.3 - https://github.com/pimoroni/pimoroni-pico/releases/tag/v0.3.3

closed 2022-01-25 by Gadgetoid