pimoroni/pimoroni-pico #247: undefined reference to `pimoroni::I2C::init()'

nymda, 2022-02-07:
I have followed the cpp install instructions and have been able to compile the examples and the boilerplate, however i cannot use any of the I2C functions due to this error.
`[build] C:\VSARM\sdk\pico\projects\pimoroni-pico/common/pimoroni_i2c.hpp:33: undefined reference to `pimoroni::I2C::init()'`

My includes are as so:
`#include "pico_explorer.hpp"
#include "bme68x.hpp"
#include "common/pimoroni_common.hpp"
#include "common/pimoroni_i2c.hpp"`

and my CMakeLists.txt file is:
`cmake_minimum_required(VERSION 3.12)

# Change your executable name to something creative!
set(NAME pico-boilerplate) # <-- Name your project/executable here!

include(pimoroni_pico_import.cmake)
include(pico_sdk_import.cmake)

# Gooey boilerplate
project(${NAME} C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

# Initialize the SDK
pico_sdk_init()

# Add your source files
add_executable(${NAME}
    main.cpp # <-- Add source files here!
)

# Include required libraries
# This assumes `pimoroni-pico` is stored alongside your project
include(libraries/pico_explorer/pico_explorer)
include(drivers/bme68x/bme68x)

# include(libraries/pico_display/pico_display)

# Don't forget to link the libraries you need!
target_link_libraries(${NAME}
    pico_explorer # <-- List libraries here!
    pico_stdlib
    hardware_i2c
    bme68x
)

# create map/bin/hex file etc.
pico_add_extra_outputs(${NAME})

# Set up files for the release packages
install(FILES
    ${CMAKE_CURRENT_BINARY_DIR}/${NAME}.uf2
    ${CMAKE_CURRENT_LIST_DIR}/README.md
    DESTINATION .
)

set(CPACK_INCLUDE_TOPLEVEL_DIRECTORY OFF)
set(CPACK_GENERATOR "ZIP" "TGZ")
include(CPack)
`

if i put pimoroni_i2c under target_link_libraries i get this issue instead:
`[build] c:/vsarm/armcc/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -lpimoroni_i2c`

sorry for asking here, but if i google "undefined reference to `pimoroni::I2C::init()'" there is literally not a single other mention of it anywhere on the internet. thanks.

Gadgetoid, 2022-02-07:
I think you're seeing the same issue that this PR fixes - https://github.com/pimoroni/pimoroni-pico/pull/246

IE: There's no library defined for `pimoroni_i2c` for external projects at the moment. I'll merge that PR and you should be able to include the new .cmake file and things will - hopefully - work.

nymda, 2022-02-07:
Thank you for the quick response, this seems to have worked :)

Gadgetoid, 2022-02-07:
It's now merged into main for future reference. I think a line linking the i2c library might be a good idea in the boilerplate too.