pimoroni/pimoroni-pico #255: Badger2040 - measure height of text

thagrol, 2022-03-01:
Can the measure_text function be extened or a new function added to return the height of the text as well as its width.

The y coordinate being the middle of the text block make positioning difficult without knowing the height.

Gadgetoid, 2022-03-01:
This is tricky. With pixel fonts we know the exact extent of the height and just return a constant, since it's impossible for any given letter to be taller than the pixel data is capable of expressing.

With these vector fonts... height is totally arbitrary. "WWW" and "www" would return different heights but should share the same baseline, and "ttt" and "jjj" might return the same height but require totally different baselines.

It's currently best to pick a vertical offset and line height that's visually pleasing and stick with it, since any sensible implementation of "height" would always be the same value for any given font/scale combination anyway.

ahnlak, 2022-07-12:
I know that this is now more of a PicoGraphics issue, rather than a Badger2040 one but... presumably the `text` function know what pixels it has set so it should be possible to get a full measurement.

I certainly understand it's going to be a more costly job to calculate, but some use cases may justify it.

(for my own purposes of partial updates, just having `text` return the affected `Rect` would be a start)

Gadgetoid, 2022-08-23:
Text measuring should hopefully be part of the complete text rendering refactor that's in progress. :eyes:

Gadgetoid, 2023-03-09:
Okay the complete text rendering refactor is still lurking out there somewhere, but more importantly Badger 2040 and Badger 2040 W now live here: https://github.com/pimoroni/badger2040/

I'm cosing this issue in favour of keeping Badger stuff in its new home!

closed 2023-03-09 by Gadgetoid