pimoroni/pimoroni-pico #272: 16 MB Pico LIPO is limited to 2MB in Micro Python?

alphanumeric007, 2022-03-04:
I'm using a 16 MB Pimoroni Pico LIPO with the custom Pimoroni uf2 file and its limiting me to the default Pi PICO 2 MB. It appears there is a need for a custom Pimoroni Pico Lipo Micro Python uf2 file, to get access to the full 16 MB?

[ https://forums.pimoroni.com/t/16-mb-pico-lipo-only-showing-4mb/19018/13 ]

helgibbons, 2022-03-10:
Just adding my datapoints to this one in case they're helpful! 

There's been some debate at Pirate Towers about whether the MicroPython file system expands automatically to fill the available flash memory. This doesn't seem to be happening for me (or if it is, some other shenanigans are afoot!) - I'm not able to upload a file that's larger than 1.3MB to my 4MB Pico LiPo, and `os.statvfs` seems to be reporting a total filesystem size of 1.375MB on both a Pico LiPo and a standard Pico.

Error when uploading a big file with Thonny:
[image]

`os.statvfs` results:
[image]
.

alphanumeric007, 2022-03-10:
I posted on the Pi Foundation Forum asking about making use of more than 2MB and was directed here.    https://micropython.org/download/?port=rp2   Listed there are custom images for the 8MB Tiny 2040 , Pico Lipo 4Mb and Pico Lipo 16MB.  I flashed the Tiny 2040 uf2 to it and was able to upload a 5.1MB mp3 file without error. The % copied actually worked too.  =) I’m guessing these images don’t have any of the files Pimoroni adds to their images? 
https://micropython.org/download/PIMORONI_TINY2040/
https://micropython.org/download/PIMORONI_PICOLIPO_4MB/
https://micropython.org/download/PIMORONI_PICOLIPO_16MB/

Gadgetoid, 2022-03-10:
> I’m guessing these images don’t have any of the files Pimoroni adds to their images?

They don't. I wonder if they could. Would save us having to build them all again.

We need to add some release GitHub Actions that build a version of MicroPython specific to each board... this has been on my "aaaaaaaaaaaaaaaaaaaaaaaah" list for a while.

Gadgetoid, 2022-03-10:
If you're logged into GitHub you should be able to find possibly maybe working builds for Pico LiPo 16MB and Tiny2040 here- https://github.com/pimoroni/pimoroni-pico/actions/runs/1966023834

Tidying up and tweaking the CI tooling to generate these, and they should be added on to our next release.

alphanumeric007, 2022-03-11:
Thanks Phil, I know your a very busy man, so taking the time to do this is greatly appreciated. Just downloaded them and will flash them in a bit. Need some more java to get fully awake. 5:30 AM here and only just had my fist sip of coffee.

Gadgetoid, 2022-03-11:
It's an overdue fix! I took the opportunity to really polish up our CI and make things faster and more efficient. With any luck I haven't broken too much :laughing: 

4MB, 8MB and 16MB builds for various boards are now available in v1.18.4 - https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.18.4

helgibbons, 2022-03-15:
Closing this one as I think we're now sorted - feel free to re-open if not :)

alphanumeric007, 2022-03-15:
Yes, sorry for not posting back, had some health issues that sidelined me a bit lately.
All sorted now, full space is available for use in Micro python, and all the needed Pimoroni uf2 files are onboard. 
A big thanks to all those who made it happen