pimoroni/pimoroni-pico #277: Badger 2040 - wrong battery reading

rdmueller, 2022-03-06:
thanx for creating the cool Badger!

I now had a chance and play round with it and I discovered, that when I attach two fresh AAA batteries, the sample code shows me an empty battery. I modified the code as described here https://cdn.learn.pimoroni.com/article/getting-started-with-badger-2040#troubleshooting , but still now useful reading.

I tried to check the vbat readings and the output seems to be 0.014, but the badger runs and the batteries are fresh.

Any idea?

helgibbons, 2022-03-09:
I'm measuring 3.2V from a pair of full AAA batteries and around 2V when they're too empty to be of use. I've tweaked the values in the tutorial (and added a link to the battery example).

Closing for now, but feel free to reopen if you're still having trouble!

closed 2022-03-09 by helgibbons

rdmueller, 2022-03-09:
Thanx

helgibbons, 2022-04-12:
Not sure which script you're referring to - if it's [the battery.py example](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/badger2040/battery.py) you can save that to your Badger (with whatever full/empty battery values you like) and run it using Thonny?

https://learn.pimoroni.com/article/getting-started-with-badger-2040#running-your-own-examples