pimoroni/pimoroni-pico #279: Website badger2040 ebook error

lurch, 2022-03-06:
https://cdn.learn.pimoroni.com/article/getting-started-with-badger-2040#customising-the-badgeros-examples says the ebook file has to be named `witw.txt`, but it looks like it actually has to be named `book.txt` ?

helgibbons, 2022-03-08:
I updated this in the tutorial last week - I think that link might be pointing to an old version. Try https://learn.pimoroni.com/article/getting-started-with-badger-2040 !

Please could you let me know where you found your link so I can update it?

lurch, 2022-03-08:
> Please could you let me know where you found your link so I can update it?

Hmmm, I'm afraid I can't remember now :confused: 
So I'll close this issue, given that the "current documentation" is correct.

closed 2022-03-08 by lurch