pimoroni/pimoroni-pico #280: Badger2040 ebook line-display bug

lurch, 2022-03-06:
If the font-size used by the e-reader fits an exact number of lines on the badger display (e.g. if three lines are visible), then every line is visible when scrolling up and down through the ebook. But if the font-size means an inexact number of lines are visible, then you can't read some lines at all!

For example if the font-size means that only 2.2 lines are visible on the display, then lines 1 and 2 will be fully visible, and only the top-part of line 3 will be visible. If you then scroll down, then lines 4 and 5 will be fully visible (and the top-part of line 6), and thus (unless you change the font-size) you don't ever get to see what lines 3 and 6 actually say :confused:

Gadgetoid, 2022-03-10:
Oof, yep, I'm seeing this on the default font size too :grimacing:

helgibbons, 2022-09-06:
Think this is now sorted, so closing for now!

closed 2022-09-06 by helgibbons