pimoroni/pimoroni-pico #281: Badger 2040 - more readable sans font?

toyg, 2022-03-06:
It's somewhat incongruous that a low-resolution device has more serif options than sans-serif ones. It should probably have one low-res-optimized typeface, i.e. a proper "pixel font". 

As it is, all sans text looks like Comic Sans, with all that it entails (I'm sure the OpenBSD crowd is very happy, but...).

Gadgetoid, 2022-03-29:
Progress:

[image]

Gadgetoid, 2022-03-30:
Bitmap fonts "bitmap6" and "bitmap8" are now available in a pre-release build here (must be logged into GitHub) - https://github.com/pimoroni/pimoroni-pico/actions/runs/2065394844

Scale must be an integer >= 1, and text is - slightly frustratingly - orientated from its top left corner. Otherwise it should work more or less the same as the vector fonts.

Will be in v1.18.6 this Friday!

helgibbons, 2022-04-07:
I've just updated the Badger docs (#337) to include the info above so I think we should be good to close this now?

closed 2022-04-07 by helgibbons