pimoroni/pimoroni-pico #286: Badger 2040 - Create Python firmware

makeworld-the-better-one, 2022-03-10:
As I understand it, right now there are two documented ways to load code on the Badger 2040:
- Use Python, and upload/edit/run code from the Thonny IDE
- Use C++, and build your own firmware and load it each time

It would be great to have a third: a way to build your own Python firmware. I can see that this is what this repo does automatically for releases using GitHub Actions, but it would be great to have some docs/instructions on how to do this with your own custom code.

This would allow for easy distribution of custom Badger programs, and easy loading of that firmware. It would also provide more code/asset space to developers, because 1. BadgerOS files wouldn't be on the disk and 2. the custom programs would be frozen / pre-compiled.

Thanks!

Gadgetoid, 2022-03-10:
We've thrown around the idea internally to make a web front-end that lets you cherry pick which drivers and builtin modules you want to roll your own totally custom build. But accepting that's probably a little too ambitious a project to handle anytime soon- yes I think build instructions are probably warranted.

It's not *too* complicated as long as you have the C++ SDK working and in the interim you might glean the proper steps from the GitHub Actions file - https://github.com/pimoroni/pimoroni-pico/blob/main/.github/workflows/micropython-badger2040.yml

It's mostly a stock-standard MicroPython build using USER_C_MODULES to include our custom drivers and also hook the install of some of the baked in files.

There's a small hack (while proper support for Badger 2040 is still pending in Pico SDK) - https://github.com/pimoroni/pimoroni-pico/blob/fe88093c300b0f73c4149fc4d1733869fd8a9857/.github/workflows/micropython-badger2040.yml#L44-L47

And all the baked in stuff is hung precariously off the USER_C_MODULES .cmake file by means of: https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/badger2040/micropython-builtins.cmake

Which is mostly just moderately complicated CMake tooling to convert assets and copy files into the right place so the MicroPython build will find them. (Albeit I think copying stuff directly into the "modules" dir might be deprecated)

That file then gets pulled in by the main modules .cmake file: https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules/badger2040-micropython.cmake

And *this* is the file that's supplied when configuring & building MicroPython.

Assuming your `micropython` and `pimoroni-pico` directories are adjacent, and you've run `git submodule update --init` everywhere (I always forget. Always), this should get you a build:

```
cd micropython/ports/rp2
make USER_C_MODULES=../../../../pimoroni-pico/micropython/modules/badger2040-micropython.cmake
```

Then you can amend the `micropython-builtins` file as you see fit.

andypiper, 2022-03-15:
I think that means the file in the paragraph immediately before, is the file that is supplied. In fact, the next section goes on to show that.

i.e. https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules/badger2040-micropython.cmake is supplied to configure and build MicroPython, as show in the commands below (`make USER_C_MODULES=...` -> that `.cmake` file just mentioned).

TheLogster, 2022-07-26:
It would appear that these directions are a little out of date. Is it possible to get an update?

andrewfstratton, 2023-01-23:
You can also use TinyGo.  I've used this myself for reading buttons, updating display, flashing the led, acting as a USB HID keyboard and serial comms over USB (to PUTTY at the moment) - all at the same time.

I would recommend you already know some Go - the TinyGo compiler and runtime are very good, but the documentation is not extensive and examples can be hard to find.

One **great** advantage of TinyGo is that you create a `uf2` file, which you can drop straight into the right device.  So you build for the badger2040 and create a uf2 that you can then share without the end users needing to install anything.  **Beware that this will remove any python/circuit python, etc.**

Also - the executable isn't big - 17k for the blinky below (without any optimisations) which is much smaller than a Go executable would be.

There's also a minimal VSCode extension that helps to set the target if you use VS Code.

FYI, here is a blinky program:
```
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})

	for {
		led.Set(!led.Get()) // toggle LED

		time.Sleep(time.Second / 4)
	}
}
```

see also https://tinygo.org/ and https://tinygo.org/docs/reference/microcontrollers/badger2040/

Gadgetoid, 2023-01-24:
> It would appear that these directions are a little out of date. Is it possible to get an update?

The most current process is always available in the GitHub Actions yaml, even though it's - I'll admit - rather hard to parse: https://github.com/pimoroni/pimoroni-pico/blob/main/.github/workflows/micropython-badger2040.yml

Due to issues with compiler versions causing MicroPython to break, your best chance of getting a working build is to fork and modify this repository and let GitHub Actions do the hard work.

I've also done a lot of work to make shipping Python files much easier in a .uf2 file.

You can pack a folder into a LittleFS filesystem and append it to an existing .uf2 firmware with - https://github.com/gadgetoid/dir2uf2

The slightly weird way Badger 2040 builds in "firmware" files will eventually be obsoleted in favour of packing a user-visible filesystem. Badger 2040W will use this method - https://github.com/pimoroni/pimoroni-pico/pull/633 - and I hope to find the time to backport it.

closed 2023-03-09 by Gadgetoid