pimoroni/pimoroni-pico #290: Badger 2040: Optimize 90 degree text rotation (or fix rotation bug?)

makeworld-the-better-one, 2022-03-11:
Having text rotation is unexpected and great, thank you! I opened this issue because I've noticed that rotating text by 90 degrees (or 180, 270, or even 360) results in wonky text. For example the `l` character isn't straight, when of course it's possible for it to be. One solution to this would be to optimize results for this 90 degree case, but the fact that even doing 360 degrees causes non-straight lines suggests to me that there is a rounding error in the text rotation code. Would be great to see that fixed! Thanks.

makeworld-the-better-one, 2022-03-11:
Maybe there's some error accumulation happening in this loop?

https://github.com/pimoroni/pimoroni-pico/blob/ab80bdde4b16cf6734299bf3d91e1e19e745a47b/libraries/badger2040/badger2040.cpp#L295-L300

If not, then maybe it's just some insufficient precision when calling `sin` and `cos` in the `glyph` method:

https://github.com/pimoroni/pimoroni-pico/blob/ab80bdde4b16cf6734299bf3d91e1e19e745a47b/libraries/badger2040/badger2040.cpp#L250-L252

I think it's the former but I'm not familiar with the codebase at all, so maybe I'm totally off base here. Let me know what you think!

Gadgetoid, 2022-03-11:
I think this is as simple as a float to int truncation, causing a value like "5.9" (which should be 6) to be truncated to 5, causing a 1px offset where there shouldn't be one.

Fix might be as simple as:

```patch
diff --git a/libraries/badger2040/badger2040.cpp b/libraries/badger2040/badger2040.cpp
index 0b65395..451fb09 100644
--- a/libraries/badger2040/badger2040.cpp
+++ b/libraries/badger2040/badger2040.cpp
@@ -264,11 +264,11 @@ namespace pimoroni {
         int8_t nx = (*pv++) * s;
         int8_t ny = (*pv++) * s;
 
-        int rcx = cx * ac - cy * as;
-        int rcy = cx * as + cy * ac;
+        int rcx = (cx * ac - cy * as) + 0.5f;^M
+        int rcy = (cx * as + cy * ac) + 0.5f;^M
 
-        int rnx = nx * ac - ny * as;
-        int rny = nx * as + ny * ac;
+        int rnx = (nx * ac - ny * as) + 0.5f;^M
+        int rny = (nx * as + ny * ac) + 0.5f;^M
 
         if(pen_down) {
           line(rcx + x, rcy + y, rnx + x, rny + y);
@@ -293,8 +293,8 @@ namespace pimoroni {
     float ac = cos(deg2rad(a));
 
     for(auto &c : message) {
-      int rcx = ox * ac;
-      int rcy = ox * as;
+      int rcx = (ox * ac) + 0.5f;^M
+      int rcy = (ox * as) + 0.5f;^M
 
       ox += glyph(c, cx + rcx, cy + rcy, s, a);
     }
```

This works for me in cases of 180 and 90 degree rotations, but I haven't exactly tested it thoroughly!

Fix should be in today's v1.18.4 release for your perusal.

makeworld-the-better-one, 2022-03-11:
Oh good idea, I forgot it would be truncating and not rounding. Thanks for such a quick response! I should be able to test later today and then hopefully close the issue.

Gadgetoid, 2022-03-11:
Or - direct link - https://github.com/pimoroni/pimoroni-pico/releases/download/v1.18.4/pimoroni-badger2040-v1.18.4-micropython.uf2

makeworld-the-better-one, 2022-03-11:
Can confirm using v1.18.4 firmware fixed this. Thank you!

closed 2022-03-11 by makeworld-the-better-one