pimoroni/pimoroni-pico #293: QR code and badge

uktechreviews, 2022-03-12:
Suggestion:

Would it be feasible to include an option qrcode.txt that automatically populates your badge photo as the QR code generated from the link?

Gadgetoid, 2022-03-14:
It might over complicate the example, but if you want to try it here's a changeset to replace the image with a QR code if you supply an additional line in badge.txt with a URL:

```patch
diff --git a/micropython/examples/badger2040/badge.py b/micropython/examples/badger2040/badge.py
index fdeae15..0d189d7 100644
--- a/micropython/examples/badger2040/badge.py
+++ b/micropython/examples/badger2040/badge.py
@@ -1,6 +1,7 @@
 import badger2040
 import machine
 import time
+import qrcode
 
 # Global Constants
 WIDTH = badger2040.WIDTH
@@ -29,7 +30,8 @@ H. Badger
 RP2040
 2MB Flash
 E ink
-296x128px"""
+296x128px
+https://pimoroni.com/badger2040"""
 
 BADGE_IMAGE = bytearray(int(IMAGE_WIDTH * HEIGHT / 8))
 
@@ -99,13 +101,38 @@ def draw_overlay(message, width, height, line_spacing, text_size):
         display.text(lines[i], (WIDTH - length) // 2, (HEIGHT // 2) + current_line, text_size)
 
 
+def measure_qr_code(size, code):
+    w, h = code.get_size()
+    module_size = int(size / w)
+    return module_size * w, module_size
+
+
+def draw_qr_code(ox, oy, size, code):
+    size, module_size = measure_qr_code(size, code)
+    display.pen(15)
+    display.rectangle(ox, oy, size, size)
+    display.pen(0)
+    for x in range(size):
+        for y in range(size):
+            if code.get_module(x, y):
+                display.rectangle(ox + x * module_size, oy + y * module_size, module_size, module_size)
+
+
 # Draw the badge, including user text
 def draw_badge():
     display.pen(0)
     display.clear()
 
-    # Draw badge image
-    display.image(BADGE_IMAGE, IMAGE_WIDTH, HEIGHT, WIDTH - IMAGE_WIDTH, 0)
+    if qrcode_url == "":
+        # Draw badge image
+        display.image(BADGE_IMAGE, IMAGE_WIDTH, HEIGHT, WIDTH - IMAGE_WIDTH, 0)
+    else:
+        display.pen(15)
+        display.rectangle(WIDTH - IMAGE_WIDTH, 0, IMAGE_WIDTH, HEIGHT)
+        size, _ = measure_qr_code(IMAGE_WIDTH, code)
+        offset_x = int(IMAGE_WIDTH / 2 - size / 2)
+        offset_y = int(HEIGHT / 2 - size / 2)
+        draw_qr_code(WIDTH - IMAGE_WIDTH + offset_x, offset_y, IMAGE_WIDTH, code)
 
     # Draw a border around the image
     display.pen(0)
@@ -193,6 +220,7 @@ detail1_title = badge.readline()  # "RP2040"
 detail1_text = badge.readline()   # "2MB Flash"
 detail2_title = badge.readline()  # "E ink"
 detail2_text = badge.readline()   # "296x128px"
+qrcode_url = badge.readline()     # https://pimoroni.com/badger2040
 
 # Truncate all of the text (except for the name as that is scaled)
 company = truncatestring(company, COMPANY_TEXT_SIZE, TEXT_WIDTH)
@@ -212,6 +240,9 @@ button_c = machine.Pin(badger2040.BUTTON_C, machine.Pin.IN, machine.Pin.PULL_DOW
 button_up = machine.Pin(badger2040.BUTTON_UP, machine.Pin.IN, machine.Pin.PULL_DOWN)
 button_down = machine.Pin(badger2040.BUTTON_DOWN, machine.Pin.IN, machine.Pin.PULL_DOWN)
 
+code = qrcode.QRCode()
+code.set_text(qrcode_url)
+
 
 # Button handling function
 def button(pin):

```

Notable issue with this is that the photo area is very small and non-square so longer URLs might result in QR codes that either don't fit or don't easily scan. But I guess URL shorteners solve that somewhat.

Gadgetoid, 2023-03-09:
Badger 2040 and Badger 2040 W now live here: https://github.com/pimoroni/badger2040/

Closing this issue to keep Badger 2040 to its new home. Please feel free to re-open over at https://github.com/pimoroni/badger2040/issues with a link back here for posterity.

closed 2023-03-09 by Gadgetoid