pimoroni/pimoroni-pico #295: Badger 2040 - Click button trigger GPIO

thalesfsp, 2022-03-13:
I would like to to connect the Badger 2040 to my RPI. Basic idea: Clicking in some Badger 2040 button would trigger an action in the RPI. Is this possible?

Gadgetoid, 2022-03-14:
Using MicroPython I'd hook it up via USB and do this over serial using Python/pyserial - https://pyserial.readthedocs.io/en/latest/pyserial_api.html

You then just need to make pressing a button print something, watch for that over serial and act accordingly.

CircuitPython could pretend to be a keyboard and send whole USB HID macros, but how useful those are depends on what you want to achieve.

thalesfsp, 2022-03-23:
I was thinking over GPIO, not serial as a keyboard.

thalesfsp, 2022-03-29:
@Gadgetoid Would be possible to connect the badger to a stepper motor driver (GPIO)?

ZodiusInfuser, 2023-01-20:
Closing this as it looks like the original question was answered, and the follow-up question was raised and answered separately https://github.com/pimoroni/pimoroni-pico/issues/323

closed 2023-01-20 by ZodiusInfuser