pimoroni/pimoroni-pico #306: Badger 2040 - use custom launcher.py if present

helgibbons, 2022-03-21:
Would be nice if `main.py` could check for a custom launcher.py before resorting to the built-in one. It currently uses the built-in version even if there a customised `launcher.py` on the Badger.

thefunkygibbon, 2022-03-24:
just change the line in main.py from 
import _launcher

to 

import launcher.py  (or whatever file you want).

just tested and that works for me

veryalien, 2022-04-02:
Or just totally overwrite main.py with a completely different script.
I modified launcher.py and loaded it as a completely different main.py directly containing all the launcher.py code.
This is totally normal micropython behaviour, it just runs main.py (if present) automatically at boot.
I agree that import _launcher is confusing and means that there is perhaps far too much 'not very useful' example code baked permenantly into the pre-built pimoroni pico micropython images.

Gadgetoid, 2023-03-09:
Badger 2040 and Badger 2040 W now live here: https://github.com/pimoroni/badger2040/

Both boards now have a user-visible "launcher.py" and none of the jankyness from the original Badger 2040 so I think this is effectively solved.

closed 2023-03-09 by Gadgetoid