pimoroni/pimoroni-pico #309: Unified display drivers

Gadgetoid, 2022-03-22:
It would be a *lot* nicer to initialize an "ST7789" class instance in MicroPython and supply a width, height and rotation instead of the current ugliness of supplying a properly sized `bytearray()`. It would also combine all of the display product libraries into a single library so we don't have duplicated functionality all over the place (getting progressively worse with every new display product :grimacing:)

## TODO

1. [x] Bind ST7789 library + PicoGraphics to MicroPython
2. [ ] Add support for various display orientations to ST7789
3. [ ] Migrate PicoDisplay, and PicoDisplay 2" and all other ST7789-based devices to the generic ST7789
4. [ ] Add shim modules (written in Python and baked into the firmware) to fake the old PicoDisplay and PicoDisplay2 module functionality
5. [ ] Update the examples and mark The Old Way as deprecated

### Phase 1 tests:

1. [ ] Pico Display
2. [x] Pico Display 2.0
3. [x] Pico Explorer

## Impact

This would change:

```python
import picodisplay as display

width = display.get_width()
height = display.get_height()

display_buffer = bytearray(width * height * 2)
display.init(display_buffer)
```

Into something a little more like:

```python
import st7789

width = 320
height = 240

display = st7789.ST7789(width, height, BG_SPI_FRONT)
```

It might involve breaking changes to the C++ API/SDK.

alphanumeric007, 2022-03-22:
I ran the following on my Pico Display (240x135 version)
Portrait & Landscape worked (swapped width and height)
Text displayed Ok both ways once I set the word wrap and made it a little smaller. It's the rainbow example. I ran the original file, which worked then modified it for the new driver.
```
# This example borrows a CircuitPython hsv_to_rgb function to cycle through some rainbows on Pico Display's screen and RGB LED . If you're into rainbows, HSV (Hue, Saturation, Value) is very useful!

import st7789
import utime

#display = st7789.ST7789(width=240, height=135, slot=0) #landscape
display = st7789.ST7789(width=135, height=240, slot=0) #portrait 

print(display)

display.set_backlight(1.0)

# From CPython Lib/colorsys.py
def hsv_to_rgb(h, s, v):
    if s == 0.0:
        return v, v, v
    i = int(h * 6.0)
    f = (h * 6.0) - i
    p = v * (1.0 - s)
    q = v * (1.0 - s * f)
    t = v * (1.0 - s * (1.0 - f))
    i = i % 6
    if i == 0:
        return v, t, p
    if i == 1:
        return q, v, p
    if i == 2:
        return p, v, t
    if i == 3:
        return p, q, v
    if i == 4:
        return t, p, v
    if i == 5:
        return v, p, q

h = 0

while True:
    h += 1
    r, g, b = [int(255 * c) for c in hsv_to_rgb(h / 360.0, 1.0, 1.0)]  # rainbow magic
    #display.set_led(r, g, b)  # Set LED to a converted HSV value
    display.set_pen(r, g, b)  # Set pen to a converted HSV value
    display.clear()           # Fill the screen with the colour
    display.set_pen(0, 0, 0)  # Set pen to black
    #display.text("pico disco!", 10, 10, 240, 6)  # landscape
    display.text("pico disco!", 10, 10, 135, 4)  # potrait 
    display.update()          # Update the display
    utime.sleep(1.0 / 60)
```

Gadgetoid, 2022-03-22:
Oof I think I need a new Pico Display then! Thank you.

I think two of my Picos are having trouble driving screens, because I switched to a third and it worked fine. Probably my slightly gooey soldering needs a touch-up but I have more imminent jobs nipping at my heels before I progress this any further.

alphanumeric007, 2022-03-22:
Happy to do any testing you want. I have a Pico Display pack V2 and V1. A 240x240 LCD breakout and a ST7735S 160x80 pixel LCD Breakout. Pico Breakout Garden and Quad pack.

Gadgetoid, 2022-05-18:
Released in beta v1.18.8 - https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.18.8

closed 2022-05-18 by Gadgetoid