pimoroni/pimoroni-pico #323: Badger 2040 and stepper motor (GPIO)

thalesfsp, 2022-03-29:
Do you if would be possible to connect a stepper motor driver to the Badger 2040 via GPIO? Any tutorial, example (wiring/code) anywhere? Thanks!

ZodiusInfuser, 2022-03-30:
If you were comfortable with soldering you could wire up to the SDA, SCL, INT, TX, RX, 3v3 and (-) on the rear for your signals and data. You would not be able to use the Qwiic/Stemma QT connector though, as this is shared with the SDA and SCL you would be using.

As for code and tutorial, there's not one I can particularly recommend, but if you search for "stepper motor with pico" that seems to return some results, such as https://www.youngwonks.com/blog/How-to-use-a-stepper-motor-with-the-Raspberry-Pi-Pico

thalesfsp, 2022-03-30:
Thank you @ZodiusInfuser! If the driver I'm planning to use, connects via SPI, would that be possible? May be, could I use some break out?

ZodiusInfuser, 2022-03-30:
Unfortunately SPI control would not be possible as the necessary pins are not exposed. If you could find a breakout that worked by I2C then that would be easy to wire up using a cable like this: https://shop.pimoroni.com/products/jst-sh-cable-qwiic-stemma-qt-compatible, but there may be difficulty with software support.

So I would say, if you are comfortable with soldering, then soldering wires to the pads on the rear that I mentioned would be the best option, as then you are dealing with direct digital IO.

Also, a thing to consider with driving a stepper is the power requirements, both in terms of voltage and of current draw. It may be a case that you have to use a separate battery pack for motor voltage, in addition to the one powering badger.

closed 2022-04-19 by ZodiusInfuser

thalesfsp, 2022-04-21:
@ZodiusInfuser Thank. you!