pimoroni/pimoroni-pico #334: Badger 2040 badger_os get_battery_level only works with USB power connected

veryalien, 2022-04-02:
In the badger 2040 example launcher.py I noticed that the battery level was only shown correctly when the USB cable was also connected. This defeats the purpose of monitoring how much battery power is left.

The ADC and VREF pins don't seem to be getting initialised correctly for badger_os, so the call to:

```
badger_os.get_battery_level()
```

doesn't return anything useful when running _only_ on batteries. Which is pretty useless.

The badger 2040 example battery.py works correctly running _only_ on battery power. That example 'initialises' the ADC and VREF pins. I copied the following to the top of launcher.py and now it's all working as I'd expect with or without a connected battery which itself is either switched on or off:

```
import badger_os
from machine import Pin, ADC

vbat_adc = ADC(badger2040.PIN_BATTERY)
vref_adc = ADC(badger2040.PIN_1V2_REF)
vref_en = Pin(badger2040.PIN_VREF_POWER)
vref_en.init(Pin.OUT)
vref_en.value(0)

# Reduce clock speed to 48MHz

...

MAX_BATTERY_VOLTAGE = 3.2
MIN_BATTERY_VOLTAGE = 2.0 
```
I think there is a least one other issue about badger battery level monitoring and which values are correct for various types of batteries. The comments in the code should be corrected and improved to clarify the min/max range for all types of batteries.

veryalien, 2022-04-02:
When I looked at the code for badger_os here:

https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/badger2040/badger_os.py

It apparantly does **exactly** the same thing, but it definitely wasn't working properly on my badger, loaded with the latest micropython release from April 1st (hmmm?!)

helgibbons, 2022-04-04:
With the new improved 'sleep' behaviour the launcher doesn't stay awake for long enough to get a true voltage reading for the battery icon - this was mentioned in the release notes for 1.18.5:

> Battery indicator mostly reads as disconnected, since it needs some time for the vref to stabilise. 

You could save the `battery.py` example to your Badger if you wanted a way to read how much battery is left when not plugged in (or mod the launcher to stay awake for longer so it can read the battery I guess, but that seems a bit counterintuitive?)

I wonder if it's worth ditching the battery icon from the launcher and adding `battery.py` as a built in if we can't think of a pleasing solution to this?

1337Reaper, 2022-04-25:
Much like the users above, I am having trouble with accurate battery readings. I have to ask what the point is of having a battery meter if it doesn't work properly? I would like to be notified that I am losing power before it actually happens.

TheLogster, 2022-07-22:
The addition of a throw-away call to  badger_os.get_battery_level() in the laucher.py is a simple fix.

Gadgetoid, 2023-03-09:
Badger 2040 and Badger 2040 W now live here: https://github.com/pimoroni/badger2040/

Closing this issue to keep Badger 2040 to its new home. I've made a note to look into this, but please feel free to re-open over at https://github.com/pimoroni/badger2040/issues with a link back here for posterity if it warrants more investigation.

closed 2023-03-09 by Gadgetoid

Arthur-Philip-Dent, 2023-10-27:
I stumbled upon this here, since I am wondering, where the battery icon is gone in [v0.0.4](https://github.com/pimoroni/badger2040/releases/tag/v0.0.4 and even )
[v0.0.3](https://github.com/pimoroni/badger2040/releases/tag/v0.0.3).

So some issue is already documentet for the 2040W [here](https://github.com/pimoroni/badger2040/issues/26) but [here](https://github.com/pimoroni/badger2040/pull/29) it says, it's not yet implemented in the badger2040. 

Did I get that right or is it just me, having no towel and being still dizzy from the improbility drive?

Thanks for any fish, err, hint! 
 
 The Hitchhiker