pimoroni/pimoroni-pico #336: Integration with Adafruit STEMMA Soil Sensor - I2C Capacitive Moisture Sensor - JST PH 2mm

thalesfsp, 2022-04-07:
Any direction on how to integrate both (wiring and coding)? https://www.adafruit.com/product/4026 Thank you!

helgibbons, 2022-04-07:
Probably easiest to use CircuitPython for this so you can follow Adafruit's guide - https://learn.adafruit.com/adafruit-stemma-soil-sensor-i2c-capacitive-moisture-sensor/python-circuitpython-test ? I don't think we have any products that use that MCU, so getting the sensor working with our libraries may not be straightforward?

You don't mention what RP2040 microcontroller you're planning on using, but we stock a few types of STEMMA/Grove cables that will fit into the connector on the soil sensor - I'd probably do some surgery on one of these so I could wire it up to the I2C pins on the microcontroller: https://shop.pimoroni.com/products/unbuckled-grove-cable-5pcs-20cm .

closed 2022-04-20 by helgibbons

thalesfsp, 2022-04-21:
@helgibbons I plan to use the Badger 2040. I bought a few STEMMA cables.