pimoroni/pimoroni-pico #338: SCD41 breakout: doesn't 'stop' cleanly in Thonny

helgibbons, 2022-04-07:
MicroPython code using SCD41 seems to run fine once but pressing stop crashes the RP2040 and requires a USB unplug/replug before the sensor will read again.

I can replicate this on a Plasma 2040 and a Badger 2040 by attempting to run the [basic example](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/breakout_scd41/scd41_demo.py) multiple times.

Gadgetoid, 2022-04-11:
Looks like it's creating a new instance of the internal object every time it's initialized. It sets an "initialized" var, but never does anything with it:

https://github.com/pimoroni/pimoroni-pico/blob/0c8b742857e88fee249fedfda7d8e71b8e99545c/micropython/modules/breakout_scd41/breakout_scd41.cpp#L47-L50

I'm guessing... or at least hoping, because most of the code involved is internal to Sensirion's driver... that this is what's going wrong.

Gadgetoid, 2022-04-11:
Okay I'm wrong- the sensor itself is getting into a bad state that the Pico doesn't like.

Removing `breakout_scd41.start()` before running the code again fixes the problem, so it seems that calling this twice is...  bad... somehow. And, indeed, calling it twice causes an immediate lockup.

helgibbons, 2022-04-11:
Thanks! Yep - doing a `breakout_scd41.stop()` before I `breakout_scd41.start()` seems to sort things!

Gadgetoid, 2022-04-11:
You're a step ahead of me :laughing: I'm baking that into the library now, along with a reinit for good measure.

Gadgetoid, 2022-04-11:
Looks like that did the trick- try the CI build from this PR: https://github.com/pimoroni/pimoroni-pico/pull/340

helgibbons, 2022-04-11:
Works great, ta!