pimoroni/pimoroni-pico #343: Micropython bindings: LCD breakout - line and triangle drawing primitives fail silently

cedel1, 2022-04-19:
When using 'display.line' and 'display.triangle' micropython methods, these fail silently (nothing get drawn on the display), while 'display.circle' works. Tested with 'BreakoutRoundLCD' class and round LCD.

**current behavior:** using 'line' and 'triangle' methods renders nothing on the display
**expected behavior:** line and/or triangle gets drawn in the display (depending on the method used)

cedel1, 2022-04-19:
I have digged into the issue a little and the result is the following pull request:

https://github.com/pimoroni/pimoroni-pico/pull/344

A few unrelated notes:
- there may be different ordering of R, G, B channels for the different micropython classes for the various displays (or it's a C/C++ or hardware difference) - cannot check that as I don't have the rectangular LCDs.
- the error seems simple enough to fix, tests should (if they existed) have discovered the problem much sooner.

cedel1, 2022-04-29:
Fixed in _main_ branch by merging https://github.com/pimoroni/pimoroni-pico/pull/344 .

helgibbons, 2022-09-06:
Closing as I think this is now merged in and sorted - yell if not :)

closed 2022-09-06 by helgibbons