pimoroni/pimoroni-pico #348: Badger 2040; Badger OS; partial_update followed by sleep seems to stop execution

jhawkesworth, 2022-05-01:
Hello,

Trying to make use of
https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/badger2040#partial-update
inside a loop, but the code seems to stop running (silently) after the first update.

I am using the 1st April release of badgerOS.

See example script below.  I was hoping just to update part of the screen, wait and then loop.  If I use partial_update(), I only ever get one update.  If I switch to update() it works fine.

I have checked and the 2nd and 4th arguments are 'multiples of 8' although not 100% sure 0 would count - the behavior is the same if I change  partial_update(16, 0, 243, 56) to partial_update(16, 8, 243, 64)

```
import random

import time
import math
import badger2040
badger = badger2040.Badger2040()

def make_random_height():
    return random.random() * 100

def make_random_temp():
    return random.random() * 100

def make_random_pressure():
    return random.random() * 10000

badger.pen(15)
badger.clear()
badger.update()
badger.pen(0)

badger.update_speed(badger2040.UPDATE_MEDIUM)

# these are junk to save re-writing some of the main code which seems to be buggy...

while True:
    badger.pen(15)
    badger.clear()
    badger.pen(0)
    
    height = make_random_height()
    t = make_random_temp()
    p = make_random_pressure()
    
    line1 = '{:3.2f}m {:3.1f}C'.format(height,t)
    line2 = '{:6.2f}Pa'.format(p)

    badger.text(line1, 16, 16)
    badger.text(line2, 16, 45)
    # PROBLEM BELOW HERE
    # badger.update() # this works fine
    badger.partial_update(16, 0, 243, 56) #- this seems to break
    time.sleep(2)
```

I notice none of the examples seem to use 'partial_update()' so maybe it should just be removed from the documentation?

Gadgetoid, 2022-05-06:
I think the error is on this line:

https://github.com/pimoroni/pimoroni-pico/blob/2413713bb8d6e4d6e715b0802d07a2db24c22af4/micropython/modules/badger2040/badger2040.cpp#L182

The `partial_update` is - by default- "blocking" and needs a "false" added here to prevent it from locking up MicroPython.

jhawkesworth, 2022-09-17:
I retried this with latest firmware on 10 September 2022 and the issue was resolved using my reproducer, so I am going to close this as fixed.  Thank you @Gadgetoid

closed 2022-09-17 by jhawkesworth