pimoroni/pimoroni-pico #349: Need Blinka version 7.2.1 or later to fix I2C pin availability for GP21/GP20 for Pico Explorer Base

barnabear, 2022-05-05:
Using the uf2 build containing Blinka 6.20.1 to program the Pico Explorer base which uses I2C on pins GP21/GP20 I get an error caused by the following issue ..

https://github.com/adafruit/Adafruit_Blinka/issues/575

The pull request is at https://github.com/adafruit/Adafruit_Blinka/pull/576

The fix is in Blinka release 7.2.1 https://github.com/adafruit/Adafruit_Blinka/releases/tag/7.2.1

Please could the Blinka version used here be updated at least to 7.2.1 to adopt this fix.

Gadgetoid, 2022-05-06:
Ooh good catch I was waiting for this to end up in a release, apparently time flies.

Pending in a PR- test build available here: https://github.com/pimoroni/pimoroni-pico/actions/runs/2281363972

barnabear, 2022-05-06:
Ooo ... great. Thank you so much. I hope to pick this up sometime this weekend and try it out.

barnabear, 2022-05-07:
Unfortunately this doesn't work because I think you either need all or none of Blinka embedded in the .uf2. It looks like you can't have part and put the rest in `/lib`

```
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
  File "board.py", line 37, in <module>
  File "adafruit_blinka/board/raspberrypi/pico.py", line 6, in <module>
  File "adafruit_blinka/microcontroller/rp2040/pin.py", line 6, in <module>
ImportError: no module named 'adafruit_blinka.microcontroller.generic_micropython'
```

My main code is adopted from an adafruit example https://github.com/adafruit/Adafruit_CircuitPython_HT16K33/blob/main/examples/ht16k33_segments_simpletest.py

... with line 19 changed to ...

```
i2c = busio.I2C(board.GP21, board.GP20)
```

I'm driving an [Adafruit 0.56" 4-Digit 7-Segment Display w/I2C Backpack](https://www.adafruit.com/product/879) from the I2C on the feature headers (not breakout garden).

Gadgetoid, 2022-05-07:
I've tried adding the "generic_micropython" dir to the modules. Right now the code that handles this is awful, but it *should* copy over just enough of a - stripped down - Blinka install to work:

https://github.com/pimoroni/pimoroni-pico/blob/323d805585b0af179f27ce7f20bb4e2865ea9c55/.github/workflows/micropython-with-blinka.yml#L63-L78

Binary should crop up here when it finishes building: https://github.com/pimoroni/pimoroni-pico/actions/runs/2287449920

barnabear, 2022-05-08:
Yes I tested this and it works in Thonny. However copying the code into main.py and replugging the pico meant that Windows was no longer able to see the USB as a valid device so the COM port was unavailable to I lost control of the device.

One step forward, two steps back.

Thanks

Gadgetoid, 2022-05-10:
I can't see why it would collapse, but it's fair to say that the Blinka/PlatformDetect compatibility layer is a little bit of a gamble so I am not surprised. Let me know if you get anywhere with it- I'm a little tied up to try it myself.

barnabear, 2022-05-20:
Thanks for that. I wonder if this would be possible for the Pimoroni Pico Lipo boards that have 4MB or 16MB flash? That would be really nice to have.

Gadgetoid, 2024-03-11:
Closing, since I don't think this is something we can support with auto builds here.

My dir2uf2 project might make it possible to build a filesystem-only uf2 for deploying Blinka though. So I'm tracking this here in case I have a sudden epiphany: https://github.com/Gadgetoid/dir2uf2/issues/6

closed 2024-03-11 by Gadgetoid