pimoroni/pimoroni-pico #351: PINS_BREAKOUT_GARDEN in Pico Explorer micropython examples are confusing

barnabear, 2022-05-08:
Some of the examples in https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/pico_explorer include the line ...

```
PINS_BREAKOUT_GARDEN = {"sda": 4, "scl": 5}
PINS_PICO_EXPLORER = {"sda": 20, "scl": 21}
```

However the breakout garden connectors on the Pico Explorer base use `{"sda": 20, "scl": 21}`. I had thought they were on a separate I2C bus, but this is not the case.

Please could the `PINS_BREAKOUT_GARDEN` lines be removed from the examples. This was quite a head scratcher for me to figure out. In fact it's all one I2C bus.

Thanks.

Gadgetoid, 2022-05-09:
That is... somewhat confusing. I had planned to move these into a library somewhere since these lines are copied all over the place :grimacing: I guess this is another vote to hurry up and do that.

ZodiusInfuser, 2022-05-09:
I believe `PICO_BREAKOUT_GARDEN` is meant to be referring to this product, rather than the breakout garden connectors on the Pico Explorer. https://shop.pimoroni.com/products/pico-breakout-garden-base?variant=32369509892179.

Also we use those pins for I2C on other products too.

ZodiusInfuser, 2022-05-09:
Oh, sorry, my mistake. Yes, those should be removed and reference the defines in this file instead: https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules_py/pimoroni.py

helgibbons, 2022-07-11:
Closing as I think this should now be sorted!

closed 2022-07-11 by helgibbons