pimoroni/pimoroni-pico #354: ESp32 with Micro Python and badger OS

robouden, 2022-05-12:
Thanks for creating the Badger OS for the badger2040. Very pleased with it on my badger2040.

I would like to know if it is possible to run the badger OS on a ESP32 with Micro Python running on it. Or if someone has done that already, to point me to the code for it.

Regards,
Rob Oudendijk

Gadgetoid, 2022-05-13:
Our MicroPython C++ drivers are incredibly RP2040 specific at the moment, so while it's not impossible it would require a fair bit of work to port the supporting C++ code.

In the case of Badger OS we're not doing much (or any IIRC) hacking on the MicroPython codebase itself, so the basic concept of a launcher and individual applications should work just fine.

robouden, 2022-05-13:
[Gadgetoid](https://github.com/Gadgetoid) Thanks for the fast reply. I will check test later to see if I can get the badger code to run on a ESP32 with Micro Phyton.

closed 2023-01-20 by ZodiusInfuser