pimoroni/pimoroni-pico #357: Trackball interrupt usage not clear

Nashenas88, 2022-05-13:
I'm writing a port of the PIM447 trackball to rust (https://github.com/nashenas88/pimoroni-trackball-driver), and while using it I'm running into an issue with interrupts. I can get the interrupts to fire, but then after about 30 seconds or so of usage, the interrupt gets stuck triggered low. I've followed the interrupt code in the C++ code closely so I imagine it has similar behavior.

While reviewing I also noticed a possible bug since get_interrupt in the trackball driver here always returns false. It's not clear to me how change_address works since the first while loop would always hit a timeout. Are interrupts supposed to work for the trackball? If so, how do I ensure they get cleared consistently?

Gadgetoid, 2022-05-14:
I believe we found a bug in the trackball firmware that affects interrupts. I'll confirm this next week and see if a fix is on the way out there.

It's nontrivial to update trackball breakouts in the field (you'd need a nuvoton programmer) so we might need to send a replacement.

Nashenas88, 2022-06-10:
Hi, any update on this? If I do need a replacement, what process would I follow?

Gadgetoid, 2022-07-22:
I had a firmware to test, but no way to flash it, and then a lot of Pico W happened. I'll see if we got anywhere with this.

Nashenas88, 2022-10-07:
@alexmaloteaux I did the equivalent in rust here https://github.com/Nashenas88/pimoroni-trackball-driver/blob/main/src/lib.rs#L339
But the issue I see is that it gets stuck low after a while. I noticed that in the code to check in interrupt near where you link (see https://github.com/pimoroni/pimoroni-pico/blob/main/drivers/trackball/trackball.cpp#L108) the code always returns false, regardless of what the value is. I wasn't sure if this meant the code wasn't expected to work.

alexmaloteaux, 2022-10-13:
@Gadgetoid any update on this ? can you release the bin firmware for those who are ok to reprogram it ?

alexmaloteaux, 2022-11-06:
i received a trackball with new firmware and it looks like the interrupt pin is fixed, i posted some capture on the pimoroni forum here : https://forums.pimoroni.com/t/pim447-trackball-how-to-get-a-clean-int-signal/17968/9?u=amalo

Nashenas88, 2022-11-06:
@PimoroniMatt is there any option to swap out old hardware with new hardware?

helgibbons, 2022-11-07:
@Nashenas88 Yep, we should be able to help with that - please drop an email to support@pimoroni.com !

closed 2023-04-26 by helgibbons

DelusionalLogic, 2023-07-07:
Would anyone happen to know which batch they were fixed in? I emailed support and got a replacement back in November (super cool of you BTW), but they seem to exhibit the same behavior of chatty interrupts.