pimoroni/pimoroni-pico #362: servo_cluster.cpp compile error?

knightinlet, 2022-05-17:
After syncing to the latest versions of the source code the build fails with errors in drivers/servo/servo_cluster.cpp complaining  error: 'is_assigned' was not declared in this scope in multiple locations within the source?

Gadgetoid, 2022-05-18:
I thought `is_assigned` was some kind of stdlib function. I'm glad I am not the only one running into this issue. I have been unable to find where it's supposed to be defined and what it does, though, but you can see from our CI that... it compile... sometimes... somehow?

I've resorted to:

```c++
#ifndef is_assigned
#define is_assigned(x) {true;}
#endif
```

I don't understand what effect `is_assigned` could ever have on a `uint8_t` anyway...

ZodiusInfuser, 2022-05-18:
Oh, this will be sub-par testing on my part! I've primarily been testing Release builds, which would compile out any assert lines. I suspect this is why the CI is fine with it too.

The actual check should be `assert(servo < pwms.get_chan_count());`, which I did in some functions it seems.

Will get a fix raised tomorrow.

ZodiusInfuser, 2022-05-18:
@knightinlet For now, try building in Release to get rid of those checks (and thus the associated errors)

knightinlet, 2022-05-18:
@Gadgetoid @ZodiusInfuser Thanks for the quick responses. Great work. Builds cleanly now :)

closed 2022-05-18 by knightinlet