pimoroni/pimoroni-pico #364: Is there a way to change the font used for rendering the badge?

MatthewSteeples, 2022-05-19:
I'm not sure why, but putting the phrase "ACTOR FOR HIRE" in the topmost section of the badge makes it difficult to read. Is there a way to change the font at all, or any reason why that combination of letters causes problems when rendering?

I know that I can pre-render my own full images and use those instead as an alternative, I was just wondering if the built in badge pieces had any options

helgibbons, 2022-05-19:
Sure! If you save your own copy of [badge.py](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/badger2040/badge.py) to your Badger you can modify it to change the fonts, sizes and thicknesses (I seem to remember reducing the thickness of the text on the 'company' line helps make it a bit more readable).

Check out the 'Running your own examples' section: https://learn.pimoroni.com/article/getting-started-with-badger-2040#running-your-own-examples

MatthewSteeples, 2022-05-22:
That's perfect, thanks. Exactly what I was after

closed 2022-05-22 by MatthewSteeples