pimoroni/pimoroni-pico #377: Pico Lipo 16mb UF2 missing PicoDisplay module(s)

RunBoris, 2022-05-29:
Pico Lipo 16mb UF2 missing PicoDisplay module(s). When trying any examples that use picodisplay or picodisplay2, ImportError: no module named...

RunBoris, 2022-05-29:
Nope, nevermind

closed 2022-05-29 by RunBoris

Gadgetoid, 2022-05-29:
I think you probably noticed from the pull request, but just to explain this -

I'm currently working on a significant number of changes which replace PicoDisplay, PicoDisplay2, PicoExplorer, 240x240 Square and 240x240 Round LCDs with a generic ST7789 driver to cover them all.

These libraries will be re-introduced to Python to contain constants defining which pins they use and no actual functionality. Similarly they will eventually be pared back to a header-only include in C++ for constants only.

If you can suffer the bleeding edge then the builds from here are the closest to how I want things to shape up: https://github.com/pimoroni/pimoroni-pico/actions/runs/2402356660

Discussion here: #373 

Usage something like this:

```python
import st7789
display = st7789.ST7789(st7789.DISPLAY_PICO_DISPLAY, rotate=0)
```

RunBoris, 2022-05-29:
Yep, I did see that. Running through the discussion on #373, it looks like
the plans are to have JPEG decoding and accompanying frame buffer to handle
loading 320 x 240 images. Is there a nightly and reference doc to test that
out?

What are the chances of GIF decoding on the fly? Is that too much for a
Pico/Micropython?