pimoroni/pimoroni-pico #380: Pico display rotate 90 degrees

nakeze, 2022-05-30:
Hello,
I am new to this product and i am implementing it in a device. From what i found i am only able to flip (180) the screen. I need to be able to rotate of 90 or 270 degrees, so that the display.text() function can show a text that is readable from the user position.
Did I miss a function ?
If it is not possible yet, do you think it is possible to add it to a next update ?

Sincerely,

Gadgetoid, 2022-05-30:
See #373

ST7789-based displays are undergoing a *huge* refactor and proper rotation support is part of this.

alphanumeric007, 2022-05-31:
It's working for me on the Display Pack V2, using that test build in the linked to thread. I actually have two Display Pack V2's side by side in patriate mode. 90 worked for me but was upside down for how I have mine orientated. I have the LED's on the bottom so I'm using 270 rotation. 
I also had it working with the Beta 1.18.8 Pimoorni uf2. 
https://github.com/pimoroni/pimoroni-pico/issues/309

alphanumeric007, 2022-06-01:
The v1.18.8 Pimoroni uf2 will get you portrait mode with the A B etc button labels right way up.
Just tested it on a Pico with Display Pack.
`
import st7789
import utime

display = st7789.ST7789(width=135, height=240, slot=0) # Display Pack portrait

display.set_backlight(1.0)

def hsv_to_rgb(h, s, v):
    if s == 0.0:
        return v, v, v
    i = int(h * 6.0)
    f = (h * 6.0) - i
    p = v * (1.0 - s)
    q = v * (1.0 - s * f)
    t = v * (1.0 - s * (1.0 - f))
    i = i % 6
    if i == 0:
        return v, t, p
    if i == 1:
        return q, v, p
    if i == 2:
        return p, v, t
    if i == 3:
        return p, q, v
    if i == 4:
        return t, p, v
    if i == 5:
        return v, p, q

h = 0

while True:
    h += 1
    r, g, b = [int(255 * c) for c in hsv_to_rgb(h / 360.0, 1.0, 1.0)]  # rainbow magic
    #display.set_led(r, g, b)  # Set LED to a converted HSV value
    display.set_pen(r, g, b)  # Set pen to a converted HSV value
    display.clear()           # Fill the screen with the colour
    display.set_pen(0, 0, 0)  # Set pen to black
    display.text("pico disco!", 10, 10, 135, 4)  # potrait 
    display.update()          # Update the display
    utime.sleep(1.0 / 60)

`

nakeze, 2022-06-02:
Thanks for this demo code ! I was able to display text like I wished.

alphanumeric007, 2022-06-02:
I was doing some testing for Phil Howard, and saved the code for future use. Glad it worked for you.

closed 2022-06-08 by nakeze