pimoroni/pimoroni-pico #383: ST7789 access to buffer to display images

nakeze, 2022-06-08:
Hello, I found a code from a previous version of the pimoroni-pico that was able to display an image : 
http://www.penguintutor.com/programming/picodisplayanimations

It is possible to update the code to make it work for the new release, just by creating the buffer and add it to the constructor : 
`display =  st7789.ST7789(width=WIDTH, height=HEIGHT, rotate180= False,buffer = loadScreenBuffer)`

I want to display a second image when the user press a button. 
I  make a secondScreenBuffer variable which is an other byteArray, but I don't find a way to "replace" the buffer of my display object.

What worked (does not seems to be the most efficient way though) is to create a second ST7789 object, and call its update function.
Did I miss a function to change the value of pointer buffer ?

Gadgetoid, 2022-06-10:
The linked experimental builds now have better JPEG support and some posted examples to get you started.

We aim to add sprite/graphics functions to make stuff like this animation example easier.

Gadgetoid, 2022-06-27:
Rotation is set at init in hardware so the graphics library can avoid the performance hit of translation.

Supplying a 320x240 buffer to a screen running in 240x320 will have very weird side-effects. You should use a smaller pen type (P4) since sharing one wont work in this case.

bisnouk, 2022-06-30:
> The linked experimental builds now have better JPEG support and some posted examples to get you started.
> 
> We aim to add sprite/graphics functions to make stuff like this animation example easier.

Hi,
does your new releasen work with C++ or only python ?
Thank's

closed 2022-07-22 by Gadgetoid