pimoroni/pimoroni-pico #386: MicroPython >= 1.8 Breaking Changes

Gadgetoid, 2022-06-10:
Currently the commit https://github.com/micropython/micropython/commit/efe23aca7154d21a9e8f283313c7a1ac29e05d86 drops the third argument from `MP_REGISTER_MODULE` which breaks compilation against our MicroPython modules. There seem to be *a lot* of changes in the pipeline, so I'm raising this issue to track breakage which need to be fixed when we bump to the next MicroPython version.

alphanumeric007, 2022-06-12:
Running your latest st7789 Pico Graphics Rewrite uf2 file my BME280 is now not working correctly? I get a reading, but it then just stays static and won't update. My BME680 is working just fine. I can pinch it with my fingers and watch the temp and humidity readings change. Nothing happens with the BME280.
My code and the micro python demo do the same thing. Stock address and 0x77. Even unplugged the BME680 while testing. It was working with an earlier official build. Not sure when it stopped working? 
The two BME280 breakouts I have, run fine in Python on my Pi 400. And in a Pico Breakout Garden running an earlier Pimoroni uf2 file. I'm pretty sure the working Pico is v1.18.7 or earlier.  
The code I'm using is as follows. Warning, wall of code follows. 
```
import time
import pimoroni_bus
import veml6075
import picographics

from machine import ADC, Pin
from pimoroni_i2c import PimoroniI2C
from breakout_bme280 import BreakoutBME280
from breakout_bme68x import BreakoutBME68X
from breakout_ltr559 import BreakoutLTR559
from breakout_rtc import BreakoutRTC
from pimoroni_bus import SPIBus
from pimoroni_bus import ParallelBus
from picographics import PicoGraphics, DISPLAY_PICO_DISPLAY_2, PEN_RGB332

vsys = ADC(29)              
charging = Pin(24, Pin.IN)
conversion_factor = 3 * 3.3 / 65535
full_battery = 4.2
empty_battery = 2.8

spibus2 = SPIBus(cs=22, dc=16, sck=18, mosi=19, bl=21 )
spibus = SPIBus(cs=17, dc=16, sck=18, mosi=19, bl=20)

display = PicoGraphics(display=DISPLAY_PICO_DISPLAY_2, rotate=270, bus=spibus, pen_type=PEN_RGB332)
display2 = PicoGraphics(display=DISPLAY_PICO_DISPLAY_2, rotate= 270, bus=spibus2, pen_type=PEN_RGB332)

display.set_backlight(1.0)
display2.set_backlight(1.0)

white = display.create_pen(255, 255, 255)
black = display.create_pen(0, 0, 0)
red = display.create_pen(255, 0, 0)
green = display.create_pen(0, 255, 0)
blue = display.create_pen(0, 0, 255)
yellow = display.create_pen(255, 255, 0)
orange = display.create_pen(255, 140, 0)
grey = display.create_pen(120, 120, 120)
violet = display.create_pen(255, 0, 255)

i2c = PimoroniI2C(sda=(4), scl=(5))
uv = veml6075.VEML6075(i2c)
connected = uv.initUV()
bme_out = BreakoutBME68X(i2c)
bme_in = BreakoutBME280(i2c,0x77)
ltr = BreakoutLTR559(i2c)
rtc = BreakoutRTC(i2c)
rtc.set_backup_switchover_mode(1)
rtc.set_24_hour()
rtc.update_time()
rtc.enable_periodic_update_interrupt(True)

while True:
    
    if rtc.read_periodic_update_interrupt_flag():
        rtc.clear_periodic_update_interrupt_flag()

        if rtc.update_time():
            rtc_date = rtc.string_date()
            rtc_time = rtc.string_time()
    
    temp_in, press_in, hum_in = bme_in.read()
    temp_out, press_out, hum_out, gas_resistance, status, gas_index, meas_index = bme_out.read() 
            
    pressuremb = press_out / 100
[... 164 lines cut ...]
        display2.text('UV:{:.0f}'.format(UVI), 140, 50, 240, 4)
          
    display2.set_pen(green)
    display2.text("??" + 'mm', 10, 120, 240, 3)
    display2.text("??" + 'mm/h', 120, 120, 240, 3)
  
    display2.set_pen(green)
    display2.text("??", 50, 190, 240, 4)
    display2.text("??", 150, 190, 240, 4)
    display2.text("??", 95, 220, 240, 4)

    voltage = vsys.read_u16() * conversion_factor
    percentage = 100 * ((voltage - empty_battery) / (full_battery - empty_battery))
    if percentage > 100:
        percentage = 100.00
    
    display2.set_pen(green)
    display2.text('battery  {:.0f}%'.format(percentage), 15, 300, 240, 3)
                 
    display2.update()
    display2.set_pen(black)
    display2.clear()
    
    time.sleep(1.0)
    ```

Gadgetoid, 2022-06-14:
We're not building against bleeding edge MicroPython just yet. This issue will come into effect when the next version is released.

closed 2022-06-17 by Gadgetoid