pimoroni/pimoroni-pico #399: Badger 2040, no prompt in Thonny

moussatat, 2022-06-22:
Hi
Correct me is this isn't the right place.
I have used the pico badger uf2 and set up the badger. I only placed badger image.  Powered up, A+C does Nothing, there is no menu.
Now, when I connect the badger and try to start micropython to modify, I will not get a prompt in Thonny.
Installing. uf2 again didn't solve the problem.
Neither did a nuke. uf2 to remove everything.
How can I reset this badger?
PS: on the other badger, I just had to use a+c to go to icon menu, then got the prompt...

Gadgetoid, 2022-06-22:
If it's the latest .uf2 you might have to go into "Run -> Select Interpreter" and pick your board manually. Thonny seems to have some trouble detecting the latest MicroPython (1.19) automatically.

alphanumeric007, 2022-06-23:
In 1.19 I'm having to select the port manually. "try to detect port automatically" doesn't seem to work? This is with a stock Pico, Pico Lipo and Tiny. I don't have a Badger. 
Once I set the port Thonny reports the board that's connected Ok. Just a FYI post.

helgibbons, 2022-09-06:
Thonny 4.0.0 seems better at auto-detecting boards running v1.19, fwiw.

Closing for now!

closed 2022-09-06 by helgibbons

Crispymug, 2024-04-04:
My Badger W doesn't appear when I attempt to select the port manually. I've tried all the troubleshooting steps I could find. How do I make it apppear?

helgibbons, 2024-04-04:
The Badger firmware lives over in this repo now: https://github.com/pimoroni/badger2040

Have you checked that you're using a known good data-capable USB cable?

If that doesn't help, I'd probably try reflashing the firmware - can you get the board to show up as a drive by holding down BOOTSEL and tapping RESET?