pimoroni/pimoroni-pico #421: Add support for Raspberry Pi Pico W

barnabear, 2022-07-02:
Please add support for the new Raspberry Pi Pico W.

I think the Pico W with the Pico Explorer base and the Breakout Garden ecosystem would be awesome.

Gadgetoid, 2022-07-04:
Pico W builds enabled as of https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.1

Build here - https://github.com/pimoroni/pimoroni-pico/releases/download/v1.19.1/pimoroni-picow-v1.19.1-micropython.uf2

Considered very beta!

And uh the "st7789" library needed for Pico Explorer has been replaced with PicoGraphics!

```
from picographics import PicoGraphics, DISPLAY_PICO_EXPLORER

graphics = PicoGraphics(DISPLAY_PICO_EXPLORER)
```

helgibbons, 2022-07-11:
Closing this one as we now have a Pico W build, yay!

closed 2022-07-11 by helgibbons