pimoroni/pimoroni-pico #423: Demo Examples  no module s7789

bsimmo, 2022-07-02:
I just had a go at running the examples following the pico explorer breakout.
Hello world works, so pico works.
Copied across custom uf2 (1.19) https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.0

In Thonny (3.3.13) with MicroPython (RPi Pico) selceted
Ran the demo example updated 19 days ago https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/pico_explorer
 and it's complaining 
```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
ImportError: no module named 'st7789'
```

alphanumeric007, 2022-07-02:
That will happen if your using the V1.19 Pimoroni uf2 file. It uses the new Pico Graphics.
Any example written for the last non beta uf2 will give you that error and need to be edited.
https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules/picographics/README.md

bsimmo, 2022-07-03:
Ah, I'm just following the Pimoroni guide and the files it points me too. There is no mention on the guide anything is beta and does not work. 
Perhaps they need to sort that out.
I'll stick it back in the draw with the other stuff that has these sorts of problems.

Not got the time to fix it myself.

alphanumeric007, 2022-07-03:
IMHO It does need some clarification and cleaning up.
If you go here, https://github.com/pimoroni/pimoroni-pico/releases
1.18.7 is labeled as the Latest.
1.18.8 is Pre Release
and 1.19 is Beta.

1.18.7 should run those examples with no editing required.

helgibbons, 2022-07-04:
I'll try and sort the outdated display examples this week, now the Pico W release chaos has eased up a bit!

bsimmo, 2022-07-04:
Thanks all, I only gave it ago (it's my son's) to see how @alphanumeric007 code was running in the forums.
That and my WeatherHAT you're advising on for me isn't working. (many thanks for that too). 

Be good to get the display side up and released for the new 1.9

Perhaps if you can put a quick change onto the shop/learn website not to use the latest code (which is what it is advising and linking to).

Got to love breaking changes to code.

Three weeks time and our school breaks up for holidays, (teacher here), so gain time to actually learn again and get this working myself.

Gadgetoid, 2022-07-04:
Sadly the breaking changes were necessary- there was no more room to handle individual libraries for every display product. MicroPython was bursting at the seems.

That said- we could have handled it better. And even now I wonder how much use an "st7789.py" would be- something to fake the old approach so users with lots of existing code written can just drop it on their Pico and carry on as normal. Oh for 72 hour days and a pill to replace sleep.

alphanumeric007, 2022-07-05:
The new Pico Graphics is in 1.19.0, and I would assume its also in the new 1.19.1. 
I have downloaded 1.19.1 but not tried it just yet.
I am running 1.19.0 and using Pico Graphics. It's working fine on my Tufty, Display Pack, Display Pack V2 and 240x240 LCD Breakout. Pi Pico and Pico Lipo, I don't have a Pico W just yet.
It's just "some of" the examples that haven't been edited to work with Pico Graphics. 

Anxiously awaiting the release of the Pico Weather Station code to see how it compares to my home brew version. And to make use of the Wind and Rain code. It's the one bit of code I've not sat down and tackled just yet.

holmesha, 2022-07-05:
Is there an easy PicoGraphics "swap" for the following piece of code in several of the examples if we are using v1.19? 

`display.set_palette_mode(st7789.PALETTE_USER)`

Or is it just easier to use v1.18.7 for now?

Gadgetoid, 2022-07-06:
@holmesha There's no `set_palette_mode` anymore, PicoGraphics uses pen types instead. See: https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/picographics#supported-graphics-modes-pen-type

You probably want something like this:

```
from picographics import PicoGraphics, DISPLAY_PICO_EXPLORER, PEN_P8

display = PicoGraphics(display=DISPLAY_PICO_EXPLORER, pen_type=PEN_P8)
```

Swapping the display as appropriate.

If you're only planning to use 16 colours you can also save ram by using `PEN_P4`.

PR #428 (2022-07-07): Update examples to use PicoGraphics
All MicroPython examples for Pico Display, Pico Explorer and the LCD breakouts are now updated to use PicoGraphics. They should now work properly with v1.19 and later (I had a general tidy up as well and renamed some examples to be more consistent).

Fixes #423 and fixes #393 .

I also fixed #408 in passing :)