pimoroni/pimoroni-pico #424: Inventor 2040 W - Where do I find the UF2 for this board.

tonygo, 2022-07-02:
Cannot locate a suitable Pico Micropython UF2 which allows the examples for Inventor 2040 W to work.

alphanumeric007, 2022-07-02:
Clicking the above "Actions" button should take you to the test builds. Then go to Micro Python. I see that there are now two Micro Python listings. I'm guessing one is for Pico and the other Pico W? That being said, that's where I linked you to in your post on the Pimoroni forums, and those images didn't work out for you? It's all I have at the moment though.
@Gadgetoid AKA Philip Howard is the guy to get you sorted out.

Gadgetoid, 2022-07-04:
I've tagged a new 1.19.1 release that includes a build for Pico W - https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.1

The file you want is : https://github.com/pimoroni/pimoroni-pico/releases/download/v1.19.1/pimoroni-picow-v1.19.1-micropython.uf2

tonygo, 2022-07-04:
Thank you. We are all making progress!

Tony

helgibbons, 2022-07-11:
Closing this one as we now have a Pico W build that will work with Inventor.