pimoroni/pimoroni-pico #425: Pico W requires a different binary file than original Pico, is there Pimoroni micropython Uf2 for the Pico W?

lesley-byte, 2022-07-02:
Does anyone have a Pimoroni flavor Micropython Uf2 for the Pico W?  I know from reading the documentation that the binary files have to be different for each board.  I suspect it might still be in development but wanted to know if there was anything out there yet.

alphanumeric007, 2022-07-02:
You could try this one, its an unofficial build, just so you know.
[https://github.com/pimoroni/pimoroni-pico/actions/runs/2594804454 ](url)

digitalurban, 2022-07-04:
Does this have the wifi modules ?

Gadgetoid, 2022-07-04:
I've just tagged a 1.19.1 pre-release that includes support for Pico W - https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.1

Modules for connecting to wireless are included, but you'll need to install various stdlib libraries to do anything useful. I've put together some examples for Pico Inky that might help - https://github.com/pimoroni/pimoroni-pico/pull/420/files

lesley-byte, 2022-07-06:
done.  Thank you

closed 2022-07-06 by lesley-byte