pimoroni/pimoroni-pico #429: any example code or driver for Pico Inky?

lesley-byte, 2022-07-06:
I've managed to read button presses on the Pico Inky via the following:

`from pimoroni import Button
import utime

button_a = Button(12)
button_b = Button(13)
button_c = Button(14)
dur = .5

while True:
    if button_a.read():
        print("Button A")
        utime.sleep(dur)
    elif button_b.read():
        print("Button B")
        utime.sleep(dur)
    elif button_c.read():
        print("Button C")
        utime.sleep(dur)
                    
`
But I would like to display on the e-ink screen as well.  Are there any libraries I should be importing...like the badgey2040 that might help?

lesley-byte, 2022-07-07:
I tried the following instead:

`from pimoroni import Button
import utime
from picographics import PicoGraphics, DISPLAY_INKY_PACK, PEN_1BIT

display = PicoGraphics(display=DISPLAY_INKY_PACK, pen_type=PEN_1BIT)

display.set_font("bitmap8")
display.text("Hello World", 0, 0, scale=2)
display.update()
button_a = Button(12)
button_b = Button(13)
button_c = Button(14)
dur = .5
message = None

while True:
    if button_a.read():        
        message = "Button A"
        print(message)
        utime.sleep(dur)
    elif button_b.read():
        message = "Button B"
        print(message)
        utime.sleep(dur)
    elif button_c.read():
        message = "Button C"
        print(message)
        utime.sleep(dur)
    message = None

`
and it only gave me a black screen but I know there's a picographics library for it.  I'm just not sure what i'm missing.  I tried to create a pen object but it caused a ValueError no matter what I did.

Gadgetoid, 2022-07-07:
You should grab this release: https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.2

And see these examples: https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/pico_inky

Sorry it's been a bit of a mad dash to fix a bug blocking these from being merged!

closed 2022-07-08 by lesley-byte