pimoroni/pimoroni-pico #437: BadgerOS API documentation

marcdraco, 2022-07-10:
I'm very new to Python, being one of those sniffy C fellows... and who needs interpreted languages anyway. ;)

The Badger is a smashing piece of hardware and I'm very impressed with the design (save for the HUGE battery clip which I've desoldered to make the thing even flatter) but I can't find anything in the way of API documentation for new programmers.

I understand OOP programming sufficiently to know what's going on but I can't find any list of the actual objects, their function or ... well anything. Thonny (unless there's something better) won't help out like say VSCode does and take me to what in C/C++ would be a header file with a bunch of defines.

Here's an example:

```
# Reduce clock speed to 48MHz
badger2040.system_speed(badger2040.SYSTEM_NORMAL)
```

Great but what other options do we have? Maybe it's my age, but I like to browse documentation to see what's around and that helps me decide if have the features I need, or, better yet avoid re-inventing a perfectly good wheel.

Anyone? Surely something exists on a project as large as BadgerOS or no one would get anything done! Maybe there's a better IDE than Thonny?

alphanumeric007, 2022-07-11:
I'm guessing you have seen this, but I'll post it anyway.
https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/badger2040#readme

Poke around in the ccp file and you may find some of the info your looking for. As far as I know the c libraries just get wrapped in Micro python.

marcdraco, 2022-07-13:
Yeah, that's about where I was. What's there is good, but it's a bit over my head as a complete noob in Python and not that strong in C++'s more advanced features.

closed 2022-07-13 by marcdraco

alphanumeric007, 2022-07-13:
I'm about in the same place. =) Self taught Python, and now Micro Python, and know nothing about C+. 
@Gadgetoid may be able to help, some? He's the most knowledgeable person I know here.