pimoroni/pimoroni-pico #448: Unable to use VL53L5CX break with Pico or PicoW (via Inventor 2040 W)

FutureMatt, 2022-07-17:
I haven't been able to get a VL53L5CX breakout to work with either a Pico W or Pico using both 1.19.2 or 1.19.1 releases.

I'm using the MicroPython examples taken from https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/breakout_vl53l5cx in both cases.

On the Inventor 2040 W running the example results in the board hanging, I then need to power cycle the board for it to work. on this board I have tried both via a soldered breakout slot and via the QW/ST connectors, both with the same result.

On a regular Pico 2040 running the example returns the following:
```
Starting up sensor...
Traceback (most recent call last):
  File "<stdin>", line 18, in <module>
RuntimeError: VL53L5CX: init error
```

In both cases I have ensured that the firmware blob is named correctly and is in the root `/`.

Gadgetoid, 2022-07-20:
Works right away for me with a Pico using 1.19.3.

Get a `init error` if I use the wrong i2c pins, so it could be a wiring or pin config issue. I'll try with a Pico W since I'd almost not be surprised at it exploding at this point (Pico W really threw a spanner in the works in a lot of ways.)

Gadgetoid, 2022-07-20:
Yup. Total meltdown on W. I strongly suspect this is because VL53L5CX is trying to allocate more RAM (on the C++/C side of things, rather than in MicroPython's pool) than is available, and has no checks in place to catch this.

Gadgetoid, 2022-07-20:
With any luck #454 should fix this. You can grab a build to test it here: https://github.com/pimoroni/pimoroni-pico/actions/runs/2706125414

MattRink, 2022-07-20:
Thanks for looking but no luck I'm afraid. It still hangs when running the demo Python code.
```
MicroPython 5bf3765 on 2022-07-20; Raspberry Pi Pico W with RP2040

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
```

MattRink, 2022-07-20:
Also you were right about the pins on the Pico 2040, it wasn't happy on 4 & 5 for some reason, switch to 20 & 21 and it runs fine.

Gadgetoid, 2022-07-21:
What's the full Python code that causes the hang in your case?

FutureMatt, 2022-07-21:
I'm just attempting to run the example here https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/breakout_vl53l5cx/vl53l5cx_demo.py.

Gadgetoid, 2022-07-21:
That's extremely odd, since that's the code I was testing against.

I'd better double check I wasn't doing something silly that made it appear to work.

Have you tried a cold boot of the sensor- fully unplugged and plugged back in again?

MattRink, 2022-07-21:
Still no luck, I also tried it via the QW/ST instead of the breakout garden connector but it made no difference (I'm trying via an Inventor 2040 W, I dont have a regular 2040 W to hand).

Gadgetoid, 2022-07-22:
Okay my local build works, but the CI build does not.

Try this .uf2 on for size:

[pico-w-vl53l5cx-fix.zip](https://github.com/pimoroni/pimoroni-pico/files/9165913/pico-w-vl53l5cx-fix.zip)

MattRink, 2022-07-22:
Great, that build works for me.

```
MicroPython v1.19.1-129-g5bf376563-dirty on 2022-07-22; Raspberry Pi Pico W with RP2040

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
Starting up sensor...
Done in 850ms...
106mm 18% (avg: 367mm 21%)
103mm 16% (avg: 363mm 20%)
102mm 16% (avg: 366mm 21%)
```

Gadgetoid, 2022-07-22:
Now I just need to figure out *why*. It could be that it's so marginal that the slightly different choices my compiler and the CI compiler, make cause a world of difference.

Trying to make it *less* marginal by reducing not-in-MicroPython-gc_heap RAM footprint. Another build incoming which will...hopefully... work.

Gadgetoid, 2022-07-22:
Okay this CI build seems to work for me - https://github.com/pimoroni/pimoroni-pico/actions/runs/2717402057

closed 2022-07-22 by Gadgetoid

MattRink, 2022-07-22:
Yup, I can confirm that the CI build works for me. Thanks for sorting it so quickly.

commit 1737023d (2022-07-22): VL53L5CX: Alloc results datai & config in MPY gc_heap.

RAM constrainits on Pico W were causing a hardlock. This change allocates the results struct, config and motion config on gc_heap to relieve system RAM pressure.

Fixes #448
commit 4a0be2f4 (2022-07-22): VL53L5CX: Alloc results datai & config in MPY gc_heap.

RAM constraints on Pico W were causing a hardlock. This change allocates the results struct, config and motion config on gc_heap to relieve system RAM pressure.

Fixes #448
commit e7a4ee02 (2022-07-22): VL53L5CX: Alloc results data & config in MPY gc_heap.

RAM constraints on Pico W were causing a hardlock. This change allocates the results struct, config and motion config on gc_heap to relieve system RAM pressure.

Fixes #448
PR #454 (2022-07-22): VL53L5CX: Alloc results data in MPY gc_heap.
RAM constrains on Pico W were causing a hardlock. This change allocates the results struct on gc_heap to relieve system RAM pressure.

Fixes #448