pimoroni/pimoroni-pico #449: REQ: time incoherent in edited Badger2040 clock.py

Martin-HiPi, 2022-07-17:
I've edited the font used by the clock.py example to "sans" and saved it back to the root of the Badger2040. It starts correctly, but it resets to the default time on every start, even if previously set. If I manually start the modified clock.py from Thonny, the time is correctly shown, confirming that the time was correctly set previously. 

Badger 2040 using [pimoroni-badger2040-v1.19.2-micropython.uf2](https://github.com/pimoroni/pimoroni-pico/releases/download/v1.19.2/pimoroni-badger2040-v1.19.2-micropython.uf2)
clock.py = https://github.com/pimoroni/pimoroni-pico/commit/b78b292ff8dadfb771f060a840cb5c64d86d8e26

Bonus question: Can the seconds be made to reset to "00" if selected? This would allow for accurate time setting :-)

alphanumeric007, 2022-07-18:
I'm not sure if your aware, but the RTC on the RP2040 doesn't have any onboard battery backup. On a stock Pi Pico it doesn't anyway. If you remove power it will reset to its default start time.
I'm thinking turning the Badger off, kills the power, and resets the time to default.
And when its being run from Thonny, its getting the time from the host PC. 

I was having a similar experience when the battery died in my RV3028 RTC breakout. It's what i'm using on a Pi Pico setup that I wanted accurate date time on.

Gadgetoid, 2022-07-18:
Indeed there is no RTC battery backup, so this cannot be fixed in software

Making seconds reset to zero is doable, though, for what little help that is.

Martin-HiPi, 2022-07-18:
I will clarify: I mean that setting the time in the clock app, pressing "B" to clear the cursor, pressing "A"+"C" to return to the launcher, and launching the clock app again, the time isn't restored in the app. It starts again from 12:00:00

A 18650 LiPo, fully charged, is attached at all times.

Since launching the clock app from Thonny shows the correct time, I know the RTC is set and time is running. Pressing "Reset" on the Badger2040 and launching the clock from the launcher fails to show the time from the RTC. At this point, the Badger2040 is still attached to the Pi. 

Once set, time is always wrong when restarted from the launcher, but good when started from Thonny. The clock app, when launched from any place, is expected to retrieve the time from the RTC and display it, but depending on where it's launched from, doesn't and this is the incoherence.

I am well aware of the RP2040 dependence on a power source to maintain the RTC, thus the attached LiPo.

Martin Boissonneault,
Technical department head
PiShop.ca|.us

Gadgetoid, 2022-07-18:
The catch is that Badger 2040 doesn't rely on the not-great sleep states of the Pico, but rather cuts the power entirely. Connected battery or not, it'll lose the time since it's truly off. The buttons actually trigger the power being turned back on, in addition to being readable by the RP2040. Rather ingenious save this one unfortunate side-effect.

Thonny quietly sets the clock in the background from the host PC.

Martin-HiPi, 2022-07-18:
I suggest this caveat be mentioned in the clock.py source then :-)

Thanks,
Martin

Gadgetoid, 2023-03-09:
Badger 2040 and Badger 2040 W now live here: https://github.com/pimoroni/badger2040/

Closing this issue to keep Badger 2040 to its new home. Have made a note to look into it.