pimoroni/pimoroni-pico #451: balls_demo.py not working

workingmatt, 2022-07-19:
I am trying to run the [balls_demo.py](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pico_explorer/balls_demo.py) on pimoroni-pico-v.1.19.2-micropython.uf2

When trying to run the code is get the following error message on line 28:
TypeError: function takes 6 positional arguments but 7 were given

This seems to echo #89 and #250.

helgibbons, 2022-07-19:
This example is working OK for me with 1.19.2 and a Pico W / Explorer Base combo - might be worth checking if an error's crept into that `for i in range:` loop?

If it's still persisting, perhaps you could post your full code and details of what hardware you're using?

workingmatt, 2022-07-19:
Thanks for the quick check. I found I made a silly error, will close the issue and shuffle off into the corner :)

closed 2022-07-19 by workingmatt