pimoroni/pimoroni-pico #459: opt gcc vs system gcc causes d_ symbols in build

Gadgetoid, 2022-07-22:
Grab ARM GCC 9, such as https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-x86_64-linux.tar.bz2?revision=108bd959-44bd-4619-9c19-26187abf5225&hash=8B0FA405733ED93B97BAD0D2C3D3F62A

Extract to `/opt/gcc-arm-none-eabi-9-2019-q4-major`

Prepend to `$PATH`

Try to build MicroPython.

Observe that a mess of runtime introspection and/or exception handling symbols appear in the build:

```
arm-none-eabi-nm --demangle=gnu-v3 --print-size --size-sort build-PICO_W/firmware.elf | grep " d_"
10079530 0000001c t d_template_args
10078a28 00000024 t d_index_template_argument.part.0
10078a4c 00000026 t d_lookup_template_argument.isra.0
10078654 0000002e t d_number_component
10078684 00000032 t d_compact_number
100784a0 00000032 t d_ref_qualifier
100787d8 00000036 t d_abi_tags
100786b8 0000003c t d_template_param
1007845c 00000044 t d_make_comp
1007a720 00000046 t d_bare_function_type
10078a74 00000048 t d_find_pack
100789d8 0000004e t d_discriminator
10079480 0000004e t d_template_arg
10078980 00000058 t d_call_offset
100794d0 0000005e t d_template_args_1
1007a768 0000006e t d_function_type
10078554 00000072 t d_append_buffer
100784d4 00000080 t d_count_templates_scopes
1007a50c 00000082 t d_parmlist
1007f428 00000082 t d_print_expr_op
10078abc 0000008a t d_growable_string_callback_adapter
100785c8 0000008c t d_number.isra.0
1007f4ac 000000a8 t d_print_subexpr
1007aa64 000000cc t d_expr_primary
100786f4 000000e4 t d_source_name
1007a3f8 00000114 t d_operator_name
10078810 00000170 t d_substitution
1007a590 0000018e t d_cv_qualifiers
1007f210 00000218 t d_print_array_type.isra.0
1007eca0 00000240 t d_print_function_type.isra.0
1007a7d8 0000028c t d_unqualified_name
1007f554 000002a8 t d_maybe_print_fold_expression.isra.0
1007eee0 00000330 t d_print_mod_list
1007f7fc 00000358 t d_demangle_callback.constprop.0
100799dc 0000035c t d_name
10079054 0000042c t d_expression_1
1007954c 00000490 t d_encoding
10078b48 0000050c t d_exprlist
10079d38 000006c0 t d_type
1007e5c8 000006d8 t d_print_mod
1007ab30 00003a98 t d_print_comp
```

The same compiler as system native is... fine?

I cannot reproduce this by building Pico Examples or the C++ module example in MicroPython, so there's something particular about our CMake files or the C++ we're using.

kilograham, 2022-07-22:
maybe look at the .DIS and see what calls them?

Gadgetoid, 2022-09-23:
The crux of the issue seemed to be `__verbose_terminate_handler` leaking in, due to the exception handling disabling being ignored.

The following snippet works around this-

```c++
namespace __gnu_cxx
{
    void __verbose_terminate_handler()
    {
    for (;;)
        ;
    }
}
```

Though why a non-distro ARM GCC has this problem when a distro supplied one does not... still eludes me.