pimoroni/pimoroni-pico #461: unicorn pack micropython does not expose .clear()

byronczimmer, 2022-07-23:
When reviewing the demos for micropython for the pico's unicorn pack, the clear function is hand rolled:
# Clear the display
for x in range(w):
    for y in range(h):
        picounicorn.set_pixel(x, y, 0, 0, 0)

The C library looks like it has a _clear() function, but it isn't exposed to micropython.

Can it be?

Gadgetoid, 2022-07-25:
Well, that's odd!

`picounicorn` *does* have a `clear()` function - https://github.com/pimoroni/pimoroni-pico/blob/0f090d8999a8c07bf00336c4602d21decab476ac/micropython/modules/pico_unicorn/pico_unicorn.c#L37

I'd venture that the examples were written before it was added... but that doesn't seem to be the case!

byronczimmer, 2022-07-26:
Not just documentation.
If I replace the hand rolled 'clear' in **micropython** with .clear() -- the display flashes ( like it does when you first start the program) but does not clear the display.

```
# Clear the display
#for x in range(w):
#    for y in range(h):
#        picounicorn.set_pixel(x, y, 0, 0, 0)

picounicorn.clear()
```

I then checked the picounicorn module in a very basic myropython script:
```
import picounicorn
print(dir(picounicorn))
>>> %Run -c $EDITOR_CONTENT
['__class__', '__name__', 'clear', 'BUTTON_A', 'BUTTON_B', 'BUTTON_X', 'BUTTON_Y', 'get_height', 'get_width', 'init', 'is_pressed', 'set_pixel', 'set_pixel_value']
```

So yes - it does exist in C, but isn't exposed to MP.  Hoping it can be!

byronczimmer, 2022-07-26:
It's been a while since I've been in C code, but should the module declaration or the picounicorn_clear_obj be '**picounicorn_clear**' and not 'picounicorn_init'. All the other module setups look syntactically similar to the object they're preparing.

https://github.com/pimoroni/pimoroni-pico/blob/0f090d8999a8c07bf00336c4602d21decab476ac/micropython/modules/pico_unicorn/pico_unicorn.c#L25

```
/***** Module Functions *****/
STATIC MP_DEFINE_CONST_FUN_OBJ_0(picounicorn_init_obj, picounicorn_init);
STATIC MP_DEFINE_CONST_FUN_OBJ_0(picounicorn_get_width_obj, picounicorn_get_width);
STATIC MP_DEFINE_CONST_FUN_OBJ_0(picounicorn_get_height_obj, picounicorn_get_height);
//STATIC MP_DEFINE_CONST_FUN_OBJ_0(picounicorn_update_obj, picounicorn_update);
STATIC MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(picounicorn_set_pixel_obj, 5, 5, picounicorn_set_pixel);
STATIC MP_DEFINE_CONST_FUN_OBJ_3(picounicorn_set_pixel_value_obj, picounicorn_set_pixel_value);
STATIC MP_DEFINE_CONST_FUN_OBJ_0(picounicorn_clear_obj, picounicorn_init);
STATIC MP_DEFINE_CONST_FUN_OBJ_1(picounicorn_is_pressed_obj, picounicorn_is_pressed);
```

Gadgetoid, 2022-07-26:
Oh wow, I glanced over the code for a bug like this and didn't see it. Good sleuthing. Thank you. You're right. I'll get this fixed!

Gadgetoid, 2022-07-26:
Test builds:

* Pico W (.uf2 not .elf) - https://github.com/pimoroni/pimoroni-pico/actions/runs/2738936946
* Non-W - https://github.com/pimoroni/pimoroni-pico/actions/runs/2738936945

byronczimmer, 2022-07-27:
Version 1.19.6 from the official downloads page for pico:

MicroPython now executes the .clear() command when called.

Thanks Gadgetoid!

helgibbons, 2022-09-06:
Sounds like this is now sorted, so closing for now!

closed 2022-09-06 by helgibbons