pimoroni/pimoroni-pico #464: partial_update freezes badger2040 with fast update_speed

joonoro, 2022-07-25:
badger2040 partial_update freezes the device if update_speed is set to anything besides UPDATE_NORMAL. the code below never prints "foo" unless speed is set to UPDATE_NORMAL. happens on v1.18.7 and 1.19.0. similar to #348 

```python
import badger2040

badger = badger2040.Badger2040()

badger.pen(15)
badger.clear()

badger.pen(0)
badger.rectangle(5, 5, 50, 50)

#badger.update_speed(badger2040.UPDATE_NORMAL) # works
#badger.update_speed(badger2040.UPDATE_MEDIUM) # doesn't work
badger.update_speed(badger2040.UPDATE_FAST) # doesn't work
#badger.update_speed(badger2040.UPDATE_TURBO) # doesn't work

badger.partial_update(5, 0, 50, 64)

print('foo')

badger.halt()
```

Gadgetoid, 2022-07-25:
Absolutely no clue why this could be the case.... none whatsoever :sob: 

```diff
--- a/micropython/modules/badger2040/badger2040.cpp
+++ b/micropython/modules/badger2040/badger2040.cpp
@@ -131,7 +131,7 @@ MICROPY_EVENT_POLL_HOOK
 
     // Ensure blocking for the minimum amount of time
     // in cases where "is_busy" is unreliable.
-    while(self->badger2040->is_busy() || absolute_time_diff_us(get_absolute_time(), t_end) > 0) {
+    while(self->badger2040->is_busy() && absolute_time_diff_us(get_absolute_time(), t_end) > 0) {
 #ifdef MICROPY_EVENT_POLL_HOOK
 MICROPY_EVENT_POLL_HOOK
 #endif
@@ -175,7 +175,7 @@ MICROPY_EVENT_POLL_HOOK
 
     // Ensure blocking for the minimum amount of time
     // in cases where "is_busy" is unreliable.
-    while(self->badger2040->is_busy() || absolute_time_diff_us(t_end, get_absolute_time()) > 0) {
+    while(self->badger2040->is_busy() && absolute_time_diff_us(t_end, get_absolute_time()) > 0) {
 #ifdef MICROPY_EVENT_POLL_HOOK
 MICROPY_EVENT_POLL_HOOK
 #endif
```

I'll, uh, raise a PR with this change momentarily.

Gadgetoid, 2022-07-25:
Okay I think that was the issue- `absolute_time_diff_us` never fails to confuse me, but this change breaks regular updates so it's not.... quite... there!

Gadgetoid, 2022-07-25:
Right, this build should be fixed - https://github.com/pimoroni/pimoroni-pico/actions/runs/2733540956

joonoro, 2022-07-25:
Yes this seems to work, thank you!

closed 2022-07-27 by Gadgetoid